carla-simulator/carla #93: Running CARLA on CPUs.

robertnishihara, 2017-12-18:
Any tips for getting CARLA running on CPUs only? When running e.g.,

```
export LIBGL_ALWAYS_SOFTWARE=1
/home/ubuntu/carla-0.7/CarlaUE4.sh /Game/Maps/Town01 -windowed -ResX=400 -ResY=300 -carla-server -carla-world-port=9988
```

on a machine with no GPUs, I see a segfault when initializing the Unreal Engine.

Or is it impossible to run Unreal without a GPU?

Full output below.

```
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(1229831809) SRandInit(1229831810).
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.082544
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin Carla
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
[... 187 lines cut ...]
[2017.12.18-19.43.43:718][  0]LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
[2017.12.18-19.43.43:718][  0]LogInit:  CurlRequestOptions (configurable via config and command line):
[2017.12.18-19.43.43:718][  0]LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
[2017.12.18-19.43.43:718][  0]LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
[2017.12.18-19.43.43:718][  0]LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
[2017.12.18-19.43.43:718][  0]LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
[2017.12.18-19.43.43:718][  0]CrashReportClientLog: Sending HTTP request: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?AppID=CrashReporter&AppVersion=4.17.2-0%2B%2B%2BUE4%2BRelease-4.17&AppEnvironment=Release&UploadType=crashreports&UserID=00021d1c0a0b151e002c01dc26fe0530%7C%7Ca8f0660a0b9d413a83ade711f8bf20cf
[2017.12.18-19.43.43:810][  0]CrashReportClientLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=1
[2017.12.18-19.43.43:810][  0]CrashReportClientLog: State change from SendingFiles to SendingFiles
[2017.12.18-19.43.43:810][  0]CrashReportClientLog: All uploads done
[2017.12.18-19.43.43:810][  0]CrashReportClientLog: State change from SendingFiles to Finished
[2017.12.18-19.43.44:715][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2017.12.18-19.43.44:715][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2017.12.18-19.43.44:748][  0]CrashReportClientLog: Final state (Receiver) = Finished
[2017.12.18-19.43.44:748][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2017.12.18-19.43.44:748][  0]LogExit: Preparing to exit.
[2017.12.18-19.43.44:749][  0]LogExit: Object subsystem successfully closed.
[2017.12.18-19.43.44:751][  0]LogModuleManager: Shutting down and abandoning module HTTP (8)
[2017.12.18-19.43.44:764][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2017.12.18-19.43.44:764][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2017.12.18-19.43.44:764][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.12.18-19.43.44:765][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

nsubiron, 2017-12-19:
The crash appears while initializing OpenGL inside Unreal Engine's code, you can try stepping into the code with GDB to see if you can get some hints, but I would say it is very unlikely to be able to run without a GPU. Since this is a general problem with Unreal Engine, you could also try asking in Unreal's forums to see if someone was able to run it on CPU only.

briancheung, 2017-12-22:
I've found that you have to run with the `-opengl3` flag. But the framerates are extremely low (~3fps) when it does run in this configuration.

robertnishihara, 2017-12-22:
@briancheung thanks for the tip!

closed 2018-01-09 by nsubiron

darld, 2020-02-27:
after adding flag, pop out window"cannot create opengl context. check thsat the drivers and hardware support at least opengl 4.3"

raviteja-kvns, 2021-02-07:
@darld were you able to figure a way out of your issue ?