carla-simulator/carla #94: Navigation

curantil, 2017-12-19:
How do the AI cars navigate in Carla?

I have checked some navigation tutorials for UE4, but it seems Carla is using something more complex.

nsubiron, 2017-12-19:
We implemented our own controller, the code can be found at "WheeledVehicleAIController.cpp". We use information about the road lanes that is generated when building the map, as well as trigger volumes for traffic lights and signs.

curantil, 2017-12-20:
Ok, if I understand it correctly, it works as follows:
- The map and basic navigation is created with the CarlaMapGenerator
- The player car (and if enough spawnpoints available, the non-player cars too) are spawned at a random PlayerStart. Those starts have to be placed manually on the map.
- Without anything else added, the cars will drive around at 20km/h with the preference at intersections of going straight, otherwise turning left.
- The SpeedLimiters can be used to easily change the speed the car is driving at (10km/h below the speedl limit)
- For random routes around the map, an IntersectionEntrance is used on which the different turning options can be created. Every route consists of one or more Actors that are the waypoints on that route over the intersection.
- Cars obey TrafficLights, but those lights need to be scripted to change color.

Is this correct?

Sometimes I get "Error: Controller doesn't have a roadmap!" and one or more cars will not move around. Any idea what could be the cause of this problem?

nsubiron, 2017-12-20:
Yes, that's basically it. The intersections are handled by IntersectionEntrance, otherwise the car just go straight until it finds a road patch with defined direction.

 A couple of comments to bear in mind for future versions
  * Cars drive 10km/h below speed limit due to a bug #5 (we want to fix this but doesn't have much priority as 10km/h below is still reasonable)
  * IntersectionEntrance will be replaced by RoutePlanner, it will be added in the next release. It's a bit more efficient and the routes are displayed with splines that looks clearer.

> Sometimes I get "Error: Controller doesn't have a roadmap!" and one or more cars will not move around. Any idea what could be the cause of this problem?

Can it be that you are dragging and dropping cars directly to the scene? Or using custom vehicles?

curantil, 2017-12-20:
> Can it be that you are dragging and dropping cars directly to the scene? Or using custom vehicles?

No. I didn't add any cars to the scene, only PlayerStarts nor have I created custom vehicles.

I also get the message : `Error: Requested 10 vehicles, but we were only able to spawn 1` although it successfully spawns 15 vehicles. (I have 18 PlayerStarts) Could it be something in the spawning process?

nsubiron, 2017-12-20:
Do you have a vehicle spawner already in the map? This actor is spawned by the game mode at begin play, you should not add a second one to the scene. The one you added doesn't have a reference to the road map thus the vehicles spawned by it are also missing this ref.

closed 2018-01-09 by nsubiron

abdoly, 2018-06-20:
Do you used RoutePlanner or IntersectionEntrance in last version ?

ishaan95, 2018-07-09:
Could you please explain why trigger volumes are used near traffic lights and signs?

abdoly, 2018-07-14:
with use of that cars understand arrive near a traffic lights and signs

nsubiron, 2018-07-15:
@ishaan95 yes, as @abdoly says. The speed limit signs and traffic lights notify this way the vehicles, and the intersections use this to detect vehicles and assign them a route.