carla-simulator/carla #96: LogCarla: Error: Missing walker spawner actor!

kikacaty, 2017-12-19:
I went into this error when I open carla via the Unreal4 editor.
I build the project from source on Linux platform following the instructions on the documentation page.

BTW, I also wonder the way to run the server and client in this setup since their is no script named `CarlaUE4.sh` if you build from source.

Thanks!

nsubiron, 2017-12-19:
Hi @kikacaty,
Yes, that error is perfectly fine. Unfortunately we cannot redistribute our pedestrians in the open project due to Adobe Fuse license terms, they are only present in the binary version. But we have left the code related to pedestrians there while we find a solution. This error complains for the lack of the pedestrian spawner, we have remove this actor too as it was too dependent on our pedestrian models.

There is no "CarlaUE4.sh" script in the source version of CARLA, you have to run from within Unreal Engine editor, just press the play button to start CARLA.

Cheers!

kikacaty, 2017-12-19:
Thanks for the quick response! @nsubiron 
I guess I can use the binary version since it seems to have some customization there.