carla-simulator/carla #102: Deserialize non-player measurements

sandman, 2017-12-20:
How can I iterate through the non-player agents' measurements in the Client? I'm not familiar with ProtoBuf. I'd like to run some checks on the non player agents and display some processed information. Thanks in advance!

nsubiron, 2017-12-20:
The protobuf objects that the API gives you are already deserialized, you can use them as any other Python object. In particular `measurements.non_player_agents` is a list containing agents

```py
for agent in measurements.non_player_agents:
    print(agent)
```

For more details, protobuf has a great [tutorial](https://developers.google.com/protocol-buffers/docs/pythontutorial), I would recommend to take a look.

sandman, 2017-12-21:
Thanks @nsubiron!
`        for agent in measurements.non_player_agents:
            if agent.HasField('vehicle'):
                tmpDist =  scipy.spatial.distance.cdist([[measurements.player_measurements.transform.location.x, \
                                                  measurements.player_measurements.transform.location.y]], \
                    [[agent.vehicle.transform.location.x,agent.vehicle.transform.location.y]])
                if tmpDist < 200:
                    print(tmpDist)`

However I do not see any prints. So I wonder if the transform coordinates are consistent for the player and the non-player agents?

nsubiron, 2017-12-21:
Distances are all in centimetres, perhaps you are using a distance too short.

sandman, 2017-12-21:
That makes sense :)

closed 2018-01-09 by nsubiron