carla-simulator/carla #97: How to get vehicle rotation matrix?

jingleFun, 2017-12-20:
what is the meaning of orientation in your proto file ? how can i get the rotation matrix? 
---------------------
"The transform contains two Vector3D objects, location and orientation. Currently, the orientation is represented as the Cartesian coordinates X, Y, Z. We will probably change this in the future to Roll, Pitch, and Yaw."

nsubiron, 2017-12-20:
Hi @jingleFun,
The orientation is defined as the forward vector of the vehicle
[image]

We are about to change it however to roll, pitch and yaw. This is already in the dev branch and will come into master in the coming release.

coolKeen, 2017-12-29:
@nsubiron @jingleFun  I don't know whether I truly understand what the orientation  means? I wander whether the coordinate of orientation is the same as the coordinate of the location?If so, when the orientation is (0.0,1.0,0.0) for example,will the car  only increase its location's y value(what I see in the carla is that the car still increase its location's x value )?

nsubiron, 2018-01-08:
If the orientation were exactly (0.0, 1.0, 0.0) the car would only move in the Y axis. But in practice the car is not perfectly oriented so you get some small variation in the X axis too.

closed 2018-01-15 by nsubiron

soulslicer, 2019-10-30:
how do i get the orinetation?