carla-simulator/carla #98: How can I train my own model under text mode ?

18605973470, 2017-12-20:
When we start Carla, a 3 dimension rendered scenario comes out. So we can intuitively get information about where the car is , what the speed is, what the front camera view is ... 
But rendering on screen is useless during training model and it costs lots of computing power.
So I want to know if there is a method to train model without rendering (e.g. under text mode)
Thanks a lot !

nsubiron, 2017-12-20:
Hi,
We have already considered this in the past, but we couldn't find an easy solution. We didn't put much effort into this because the performance gain is probably small, (the main thing slowing down the render are the scene captures not the main camera). One hacky thing you can do is reducing its resolution at minimum

    $ ./CarlaUE4.sh -windowed -resX=5 -resY=5

However, if you find a nicer way to deactivating the main camera we'll be very happy to hear about it :)

nina124, 2017-12-26:
Hi, I am new to this environment. Two questions troubled me a lot.
I log in to my lab server via `ssh server -X` then is it possible to build Carla under this condition?
I have this question because  `./Rebuild.sh` uses opengl to render, but `glxinfo` fails. Below is the `glxinfo` output.
> name of display: localhost:11.0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  45
  Current serial number in output stream:  46

Another question.
Assume I have built carla successfully on the lab server. When training with it, can I start the carla server if I log in to the ubuntu server via `ssh server -X` ?
If not, how to do that?

If answers to the above two questions are neither 'Yes', I think the render necessity brings too much incovenience, and strongly suggest  developers to remove that.

Thanks a lot.

closed 2018-01-09 by nsubiron