carla-simulator/carla #99: Creating Carla Map Generator causes crash

curantil, 2017-12-20:
When I try to add a "Carla Map Generator" to an empty level, the level editor crashes.

This is the first error message that appears (longer log is available if it helps):
```
[2017.12.20-12.19.47:049][ 29]LogEditor: Attempting to add actor of class 'CarlaMapGenerator' to level at 320.00,-310.00,100.00
Using fixed seed 123456789
[2017.12.20-12.19.47:050][ 29]LogCarla: Generated DCEL with: { 12 vertices, 32 half-edges, 6 faces }
Fatal error: [File:/Storage/wim/Projects/UnrealEngine_4.17/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLVertexDeclaration.cpp] [Line: 91] 
Unknown RHI vertex element type 0
Signal 11 caught.
```

This seems to be an [UE4 4.17.2 issue](https://answers.unrealengine.com/questions/709850/crash-in-4172-with-instanced-static-meshes-unknown.html). But what is the recommendation for this as the recommended UE4 version for Carla is 4.17.2?

nsubiron, 2017-12-20:
Hi @curantil,
Are you on Linux or Windows? I cannot reproduce in my machine with 4.17.2 on Linux. Perhaps you can downgrade to 4.17.1 until we upgrade CARLA to 4.18. We have a Windows machine with 4.17.1 and it's working without issues.

curantil, 2017-12-20:
I'm working on Ubuntu 16.04. I can create a map by duplicating an existing map and reusing the CarlaMapGenerator, but not by placing a new one in an empty map.

tanisxu, 2018-01-05:
@curantil excuse me, are you working on generating new maps in carla? would you please show me how exactly you do this job? I can't find where the document shows the method.In http://carla.readthedocs.io/en/latest/map_customization/ I can not understand what they try to explain.
Thank you and happy new year.

tanisxu, 2018-01-05:
By customization I mean use script like python or sth to generate new map.

curantil, 2018-01-05:
I have put my answer in your question #110

nsubiron, 2018-02-16:
I am able to reproduce this issue now, same machine but with a fresh installation of Ubuntu 16.04.

Since it seems to be an issue of the Instanced Static Mesh and soon we are planing to change the method the road map generates the road, hopefully it will stop being an issue soon.

PR #275 (2018-03-15): Optimizations
#### Description

Introduce a set of changes that will add optimizations into the CARLA Unreal Engine project. Initially includes Low and Epic quality settings levels.

Changes applied in this branch:

  - Generates pieces for the road
  - Add CARLA settings page in the Unreal Project Settings screen:
  - Draw distance for low quality levels for light, road pieces and other primitive components

###### Fixes

- Fixes #253 Add quality level settings
- Fixes #252 Use actors instead of instances in road generator
- Fixes #251 Screen captures are now being performed inside the render [... cut]
closed 2018-03-15 by nsubiron