carla-simulator/carla #103: Bug!!!make: *** No rule to make target `CarlaUE4Editor'.  Stop.

yiakwy, 2017-12-21:
I am currently build it on MacOS and CentOS. Everything is good now after I tackled the dependency and Unreal app problems. But one last thing prevent me from proceeding:

> There is no Makefile in Unreal/CarlaUE4. And make will throw bugs below

```bash
$UE4Editor $PWD/CarlaUE4.uproject
# lack two .dylib files I think it might be compiled to .so so that MacOS users cannot use it
# then I proceed to Makefile
# but there is no Makefile and make simply throws error
# make: *** No rule to make target `CarlaUE4Editor'.  Stop.
```

Where is the "makefile" in the folder?

nsubiron, 2017-12-21:
Hi @yiakwy, "Rebuild.sh" is meant for Linux, we don't have support for MacOS or CentOS. 

Compiling for MacOS is most probably possible, however you'll have to do some manual work. Please check the instruction to build [Unreal Engine from source](https://docs.unrealengine.com/latest/INT/Programming/Development/BuildingUnrealEngine/index.html).

closed 2017-12-21 by nsubiron

yiakwy, 2017-12-21:
Hello@nsubiron, thanks for your reply. I did compiled Unreal Engine from source and I can run it from Xcode.  I think that there should be a Makefile inside Unreal/Carla. I can generate xcode projects file, does it mean that I can build it inside xcode? I can specify Makefile dependency inside Xcode project.

(1) So the first question, could you show me what the Makefile for Unreal/Carla looks like? Which program generates it?

 

I got an error threw Source/Carla/SeceneCaptureCamera.cpp Line 27

```objectiveC
static constexpr auto DEPTH_MAT_PATH =
#if PLATFORM_LINUX
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial_GLSL.DepthEffectMaterial_GLSL'");
#elif PLATFORM_WINDOWS
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial.DepthEffectMaterial'");
#else
#  error No depth material defined for this platform //Line 27
#endif

```
And another error from the same file, Line 30

```
static constexpr auto SEMANTIC_SEGMENTATION_MAT_PATH =
    TEXT("Material'/Carla/PostProcessingMaterials/GTMaterial.GTMaterial'");
```

xcode complains that "Expected expression" at the first token "static".

Could you give me any suggestion to tackle those problems?

nsubiron, 2017-12-21:
Hi,

The Makefile is generated by Unreal Build Tool and it is specific for Linux, in Windows for instance it uses Visual Studio instead of Makefiles. In MacOS I think you build from Xcode, but I have never tried.

And regarding the error, we put it this way because the depth map shader is dependent on the graphics API and we have never tested it in MacOS. You can use one of the two and test, just replace those lines by
```cpp
static constexpr auto DEPTH_MAT_PATH =
#if PLATFORM_LINUX
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial_GLSL.DepthEffectMaterial_GLSL'");
#else
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial.DepthEffectMaterial'");
#endif
```

yiakwy, 2017-12-21:
@nsubiron that works for moment. And change the default clang c++ gramma to c++1z. errors eliminated. 

I didn't build llvm@3.9 locally and use MacOS "brew" to install llvm@3.9 followed by making soft links to clang  at /usr/local/bin.

However, I checked brew formula that macOS llvm@3.9 didn't ship libc++abi component hence I don't have libc++abi.a which stops compilation.

For example in my install_boost_locally_macos.sh (I will source it within Setup.sh), I have
```bash
pushd boost-source >/dev/null

LDFLAGS="-L/usr/local/opt/llvm@3.9/lib -Wl,-rpath,/usr/local/opt/llvm@3.9/lib"
CPPFLAGS="-I/usr/local/opt/llvm@3.9/include"
PY_SYS_PREFIX=$(python-config --exec-prefix)
PYFLAGS="-I${PY_SYS_PREFIX}/include/python3.6m"
LDPYLIB="-L${PY_SYS_PREFIX}/lib"
export PATH="/usr/local/opt/llvm@3.9/bin:$PATH"

BOOST_TOOLSET="clang-3.9"
BOOST_CFLAGS="-fPIC -std=c++1z -stdlib=libc++ ${CPPFLAGS} ${PYFLAGS}"
BOOST_LFLAGS="-stdlib=libc++ ${LDFLAGS} ${LDPYLIB}"

./bootstrap.sh --with-toolset=clang --prefix=../boost-install
./b2 clean
./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" linkflags="${BOOST_LFLAGS}" --prefix="../boost-install" -j 4 stage release
./b2 install toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" linkflags="${BOOST_LFLAGS}" --prefix="../boost-install"
```
It works fine (I am boost users, and I am sure it works well with brew llvm). And I did the similar things for all other dependencies. They all are built locally and look good.

I have following libraries inside plugins
```bash
(Py3) ➜  CarlaUE4 git:(master) ✗ ls Plugins/Carla/CarlaServer/lib/
libboost_system.a libcarlaserver.a  libcarlaserverd.a libprotobuf.a
```

Now Xcode complains that 
```bash
Showing Recent Issues
clang: error: no such file or directory: '/Users/lei.wang1/Github/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libc++abi.a'
```

Why libc++abi.a is needed? Is there any workarround

nsubiron, 2017-12-21:
I'm not 100% sure that c++abi is needed, you can try removing it from the Carla plugin dependencies, and see if it compiles without link errors. Just remove the line 111 in the "Carla.Build.cs" file

https://github.com/carla-simulator/carla/blob/9855fe35f9691a4e9346f6a4525de08777c9cbf9/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs#L111

This file sets up the all the dependencies of the Carla plugin and it's used by the Unreal Build Tool to generate and compile the project.

yiakwy, 2017-12-21:
@nsubiron  Build successfully! Even though the editor emit warning. "Mesh" are found. Another suggestion is, we can build Unity separately and build Carla afterward. Hence argument "-engine" is not needed.

nsubiron, 2017-12-21:
The "-engine" argument is there mostly because is the recommended way of building in the Unreal documentation for Linux. Since "make" makes sure only out-of-date files are recompile the engine builds only once. Plus it might come handy when you need to rebuild the engine sources; from time to time after some Linux upgrade you get this nasty error

    fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header '..../SharedPCH.Engine.h.gch' was built

then you need to recompile also engine files, you can easily do it with

    $ make CarlaUE4Editor ARGS=-clean
    $ make CarlaUE4Editor

takes a lot of time but as far as I know it's necessary.

yiakwy, 2017-12-22:
Get it! Thank you @nsubiron .