carla-simulator/carla #106: steer angle question？

coolKeen, 2017-12-24:
In the document.I see that steer angle ranges from [-1,1] and the description like this ：
“The actual steering angle depends on the vehicle used. The default Mustang has a maximum steering angle of 70 degrees (this can be checked in the vehicle's front wheel blueprint).”
Dose it mean -1 represents -70 degrees (negative value makes the car move to left) and 1 represents 70 degrees(positive value makes the car move to right)?
Thanks a lot~

n-akram, 2017-12-24:
Hi,

By referring to general car mechanics and other simulation engines, I infer that 70 degree should mean the range. In a way that -1 would correspond to -35 degrees and +1 would correspond to +35 degrees.

coolKeen, 2017-12-24:
@n-akram Thanks。Whether that means  in carlar, if I want to use 3 cameras like NVIDIA's [paper](https://arxiv.org/abs/1604.07316) ,I should not make the left camea and right camera diverge the center camera by 35 degree for the reason that I need to give left image and right image the correct throttle but I only have the center image's throttle and the throttle can only   range from -35 to 35 degree ?

n-akram, 2017-12-24:
@coolKeen, according to Nvidia paper, three cameras are placed behind windshield. If you are trying to recreate that in CARLA, just positioning three cameras at correct position is enough. Cameras will move with car once positioned. You dont have to explicitly define  its movement.

Steering left and right will not directly effect camera orientation, as it is rigidly attached to car. Camera orientation will change only with car.

Throttle does not range between -35 and 35 degrees. Throttle ranges between 0 and +1 for forward and 0 and -1 for reverse.

coolKeen, 2017-12-24:
@n-akram Thanks for respond! I am  sorry that I make a typo on my question. What I  really want to ask is that if I want to give left image and right image the correct steer(instead of throttle),should I need to make the left camera and right camera diverge the center camera by 35 degree.I need to give left image and right image  correct steer and throttle because I want to do supervised learning,but the problem is that I only have the center image's steer and throttle,so I need to calculate the suitable steer for left image and right image。I don’t mean that I want to explicitly define their movement when car runs,what I mean is that I need to give left image and right image  "label"(i.e.steer and throttle) in my trainning code.

n-akram, 2017-12-24:
@coolKeen if I understood correctly, you want to use images from right, left and centre cameras with steering angle and throttle as label.

If you consider centre camera images, steering angle and throttle as one set, images from the left and right camera also would have same steering angle and throttle values. You do not need to change anything.

dosovits, 2017-12-24:
@coolKeen , indeed you need to adjust the steering for the left and right images, but the angle you adjust it by does not have to be the same as the angle the camera is rotated by. These are very different things - camera rotation angle and steering angle, they do not have to be equal.

n-akram, 2017-12-24:
@coolKeen as @dosovits said, you need to make adjustment if you are augmenting the data. I was under the impression that you are training using three camera images for the same variable. 

However, adjustment would depend on speed, camera position etc and it is not same as steering angle.

coolKeen, 2017-12-25:
@n-akram @dosovits  Thanks! By the way, any suggestion on how to adjust the steer and throttle for left and right images?

n-akram, 2017-12-25:
Hi @coolKeen , I have not given it much thought. I think you can place three cameras on an arc. That way, you can take the change in orientation of the vehicle is proportional to steering angle (for small distances). 

For Throttle, I think you could assume engine speed in rpm proportional to throttle (between minimum and maximum engine speed). And then you can use the vehicle dynamic formulas to obtain the distance travelled. In your case, you would know the distance travelled, and you will have to obtain engine speed in rpm. You can use this to obtain throttle (will be -ve for reverse).

coolKeen, 2017-12-26:
@n-akram Thanks

nsubiron, 2018-01-08:
It's actually a bit more complex, 70 is the maximum angle that a wheel is set up to turn, but bear in mind that the steering angle is different for each wheel when making a turn. The outer wheel needs to travel more distance than the inner wheel therefore has a lower steering angle.

This is all calculated by the PhysX package, measuring:
 
[image]

coolKeen, 2018-01-08:
@nsubiron Thanks for respond!By the way,could you tell me how you adjust the steer and throttle for left and right images when using left camera and right camera in your paper "End-to-end Driving via Conditional Imitation Learning"?

nsubiron, 2018-01-08:
@coolKeen I don't understand what you mean with "adjust the steer and throttle for left and right images".

The cameras are rigidly attached to the vehicle. Since you know the vehicle's pose at every frame you can calculate the cameras movement if that's what you need. Steer and throttle values are the input you give to the simulated engine of the vehicle, the resulting net movement depends on many things like engine parameters, vehicle weight, inclination of the terrain, etc.

closed 2018-01-15 by nsubiron

surendrasah, 2020-01-24:
Hi,
I have steering angles but its in positive sign values like 20, 30, 120 etc. degree. how would i calculate if it turns left or right. CARLA onöy expects value betwwen -1 to 1. 
 Any idea on this would be really appreciated.

rehamessameltagoury, 2020-05-05:
Hi 
I am trying to implement this model but the steering predictions are not right.  
So i wanted to know if this model worked with you

rehamessameltagoury, 2020-05-05:
Hi 
I am trying to implement this model but the steering predictions are not right.  
So i wanted to know if this model worked with you😅

wangzhensuo, 2021-02-05:
steer angle range  -35 to 35 ,it is not correct.   -70 to 70 is correct.
below is the msg in /vehicle_info in autoware.
max_steer_angle: 1.22173035145   is radian 。
transfer to angle is 70.

type: "vehicle.lincoln.mkz2017"
rolename: "ego_vehicle"
wheels: 
  - 
    tire_friction: 3.5
    damping_rate: 0.25
    max_steer_angle: 1.22173035145
  - 
    tire_friction: 3.5
    damping_rate: 0.25
    max_steer_angle: 1.22173035145
  - 
    tire_friction: 3.5
    damping_rate: 0.25
    max_steer_angle: 0.0
  - 
    tire_friction: 3.5
    damping_rate: 0.25
    max_steer_angle: 0.0

GooVincent, 2021-11-16:
run following code, will obtain the current vehicle max steer angle in Carla.
`physics_control = vehicle.get_physics_control()
physics_control.wheels[0].max_steer_angle
physics_control.wheels[1].max_steer_angle
physics_control.wheels[2].max_steer_angle
physics_control.wheels[3].max_steer_angle`