carla-simulator/carla #107: Help requested: Crash on libGL loading

gilamid, 2017-12-25:
Hi, 
I am looking for help - trying to run the compiled version of Carla.
seems like It is crashing on loading the openGL library

I have also tried the work around suggested in:
 #93 
opened by @robertnishihara

but the crash is still there. 
Am i doing something really stupid here - or is there an issue ?

Any help will be appreciated.
Thanks !

Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(0) SRandInit(0).
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.176037
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin Carla
[33mLogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
[0mLogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
[... 190 lines cut ...]
[2017.12.25-09.02.05:722][  0]LogInit:      CURL_VERSION_TLSAUTH_SRP
[2017.12.25-09.02.05:722][  0]LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
[2017.12.25-09.02.05:722][  0]LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
[2017.12.25-09.02.05:723][  0]LogInit:  CurlRequestOptions (configurable via config and command line):
[2017.12.25-09.02.05:723][  0]LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
[2017.12.25-09.02.05:723][  0]LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
[2017.12.25-09.02.05:723][  0]LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
[2017.12.25-09.02.05:723][  0]LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
[2017.12.25-09.02.05:723][  0]CrashReportClientLog: Sending HTTP request: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?AppID=CrashReporter&AppVersion=4.17.2-0%2B%2B%2BUE4%2BRelease-4.17&AppEnvironment=Release&UploadType=crashreports&UserID=0002306d0c0c0a19001b00daeed5ab9e%7C%7Cf940e0425e01688dc044308a5a37133a
[2017.12.25-09.02.06:460][  0]CrashReportClientLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=1
[2017.12.25-09.02.06:460][  0]CrashReportClientLog: State change from SendingFiles to SendingFiles
[2017.12.25-09.02.06:460][  0]CrashReportClientLog: All uploads done
[2017.12.25-09.02.06:460][  0]CrashReportClientLog: State change from SendingFiles to Finished
[2017.12.25-09.02.06:741][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2017.12.25-09.02.06:741][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2017.12.25-09.02.06:741][  0]CrashReportClientLog: Final state (Receiver) = Finished
[2017.12.25-09.02.06:741][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2017.12.25-09.02.06:741][  0]LogExit: Preparing to exit.
[2017.12.25-09.02.06:742][  0]LogExit: Object subsystem successfully closed.
[2017.12.25-09.02.06:744][  0]LogModuleManager: Shutting down and abandoning module HTTP (8)
[2017.12.25-09.02.06:758][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2017.12.25-09.02.06:758][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2017.12.25-09.02.06:758][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.12.25-09.02.06:758][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

felipecode, 2017-12-30:
Hey @gilamid 
We are on vacation until the 7th. We are definitely looking at this when we come
back.
Cheers and Happy new year.

ChongYunPeng, 2018-01-06:
Hi,

I have the same issue. When I am accessing my Ubuntu computer with multiple GPU, I am not able to run CarlaUE4.sh.

It seems to work when I run it via ssh, though I don't get any display.

I have Unreal Engine installed but I assume that is not needed for the compiled version.

closed 2018-01-15 by nsubiron

tartavull, 2018-04-02:
Same problem when using nvidia-docker.

SunHaoOne, 2021-03-10:
I solve this problem using nvidia-docker, but it shows me Waiting for the client to connect..., I do not know how to figure it