carla-simulator/carla #108: autopilot question？

coolKeen, 2017-12-27:
Hi，I run autopilot mode in client_example.py  but finding that the throttle value change only between three value（ 0.0 ，0.5 and1）。I want to know why？Thanks a lot！

felipecode, 2017-12-30:
Hey @coolKeen  sorry for late answer.
The contributors are currently on vacation until the 7th.
This is something arbitrary. The agents are quite simple, they just go for the speed
limit of a certain zone, and then they keep this speed. In a proportional controller style.

A throttle of 1 increases the speed. So this value is used when the autopilot wants to increase speed
A throttle of 0.5, more or less maintain the same speed.
Without throttle the car reduces the speed.

Happy new year.
Cheers.

coolKeen, 2017-12-31:
@felipecode  Thanks for respond！I am so sorry to disturb your holiday，but  my recent experiment is a bit urgent。I have three more question。
1.Does the  the steer value -1 represent -35 degree（make the car left） and 1 represent 35 degree（make the car right）？（I have asked this question in #106 ,just for confirm）
2.I don't know whether I truly understand what the orientation means? I wander whether the coordinate of orientation is the same as the coordinate of the location?If so, when the orientation is (0.0,1.0,0.0) for example,will the car only increase its location's y value(what I see in the carla is that the car still increase its location's x value )?( in #97)
3.How can I get the current speed limit？
Again，So sorry to disturb your holiday.

Happy new year.
Cheers.

coolKeen, 2018-01-08:
@felipecode

nsubiron, 2018-01-08:
Hi @coolKeen, sorry for the late response.
1. The steer angle is different for each wheel, the inner wheel steers (-)70 when the steer value is (-)1. See https://github.com/carla-simulator/carla/issues/106#issuecomment-355985118.
2. You are right about the orientation, if the car was oriented on the Y axis only the Y coordinate of the location would change, but in practice the car is never perfectly oriented so you see small variations in the X coordinate too.
3. The speed limit is not yet available at the client side, sorry. You know however where the speed limit signs are located in the map as they listed in the list of non-player agents that is sent to the client, and from that you can guess the current speed limit.

Cheers!

coolKeen, 2018-01-08:
@nsubiron  Thanks for respond!!!

closed 2018-01-09 by nsubiron