carla-simulator/carla #110: How can i customize my map ？

tanisxu, 2018-01-04:
I appreciate this project much， in your design you‘ve consider such things as lidar ，vehicle spawned and map edit， awesome.
But when I read docs about map customization in:
http://carla.readthedocs.io/en/latest/map_customization/  
and
http://carla.readthedocs.io/en/latest/how_to_add_assets/#map-generation  
I still confused about how exactly I should do to get my own map?
Other wise, I notice that the road contains only yellow lane_line,  and roads are all two lanes, but in real world there're many white lines, and roads may contains three or more lanes, will you do something about it?
Looking forward for your answear and thank you for developing such an awesome project.

tanisxu, 2018-01-05:
By customization I mean use script like python or sth to generate new map.

tanisxu, 2018-01-05:
@curantil Thank you !!! I'll spend some time looking into it.

tanisxu, 2018-01-06:
@curantil I understand change the seed will make roads different. But how can we generate roads as we need(eg, we can get data which is a real map and need to generate a ue4 map the same as real world).
Is there something we can do with SplineMeshRepeater?

TheNihilisticRobot, 2018-01-08:
Hi @tanisxu and @curantil We're back from winter break.

Everything @curantil said is right. Thanks for covering us during vacation!

As for your unanswered questions, @tanisxu, there is not inmediate plan of doing a script to get map information and transforming it into a Carla level, though we might reconsider it in the future. We plan to upgrade the Map generator allowing more customization and more organic layouts (splines might be implied, more lane configurations, different intersections...) but is a complicated task and We'll be covering other issues before those. The closest thing to come related to your questions will be a set of tools to make easy to adapt any custom level or Unreal store map fully usable for Carla.

closed 2018-02-01 by carla-admin

chengjinjack, 2018-09-25:
hello,@curantil
I  don't know that how to place  IntersectionEntrances to the level.
Hope to get your help.

TheNihilisticRobot, 2018-09-25:
You are most likely looking for RoutePlanners, as every other asset you can find it by searching for its name on the Mode Window "place" Mode.

rjtshrm94, 2020-01-24:
Hi, I did every thing as mentioned by @curantil  but When i play the level in unreal engine it automatically undo the changes and restores to the same level as earlier. Steps I did was:
1. Duplicate the existing map
2. Changes to the map (removing assets and keeping roads only)
3. Save current
4. Build and again save
5. Play
but cant see the new changes. Same as the old map.

curantil, 2020-01-24:
That description was for an earlier version of Carla (0.8.x?). The 0.9.x version is quite different, so I think most of the explanation will not work unless you use the old version. 

The documentation of the map making page for modifying an existing map in the 0.9.x version seems to be a dead link https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/

A way to generate a complete new map is using a tool called RoadRunner, but sadly that tool is not free. (well you can get a time limited trial which you can use for two weeks if I'm not mistaken. But that gives you hardly time to do it properly.)

abdellatifzaarour, 2025-05-08:
hello evryone i am just an amateyre in 3d designe how can i designe a long straight highway for my  outonoumous test