carla-simulator/carla #112: Carla vs Airsim

tanisxu, 2018-01-05:
I've notice another simulator named airsim. After some basic installation and use, I found it have better grapic effect than carla, but it seems airsim is weak in controlling other car, and no red light provided either. Also airsim does not have such a good roadmap as carla do.
Anyone there who are familiar with both carla and airsim? Can someone tells the disadvantage or advantage of the two framework?
Thank you guys and wish you good luck in the new year,

curantil, 2018-01-05:
If there is a difference in graphics it is probably due to the use of different assets, so they might be interchangeable.

Both are still heavily under development, but for me, the ground truth segmentation is more flexible in AirSim. On the other hand, the advantages of Carla are the weather system and the AI (if you have enough with a grid based road pattern). There probably are more advantages between both, but those are the ones I came across.

tanisxu, 2018-01-05:
@curantil Thank you. 
By the way, what's your opinion on airsim and carla's map building system? which one is easier to use under python script or cpp generation codes?
I'm new to both plugin and feel a liittle confused.
Thanks for your  explaination.

curantil, 2018-01-05:
I have not done any map building for Airsim yet. Although as far as I understand it doesn't have an automatic generator like Carla. Although, if you need more than a grid of roads, you basically fall back on the ue4 editor for both although you might repurpose some elements of Carla. 

Both simulators provide a c++ as well as a python interface, but I haven't enough experience to compare them.

tanisxu, 2018-01-06:
@curantil Thanks a lot !

TheNihilisticRobot, 2018-01-08:
Hi @tanisxu @curantil 

The map you see in AirSim is an asset from the [Unreal Marketplace. ](https://www.unrealengine.com/marketplace/modular-neighborhood-pack)It is a great  and flexible pack of assets. We couldn't add it to the released project because of the terms and conditions of the Marketplace (It would be redistribution). If you wish to use it with Carla you may buy it and add it to your project.  Though you won't have the road lane information, the street signs and the autopilot will not work yet, we'll make possible to add all Carla features to any map you find/make including the one they use for airsim.

xie-dongping, 2018-01-14:
AirSim does not support stepping yet, however, this is essential if one is to run high fidelity models are that not real time capable.

https://github.com/Microsoft/AirSim/issues/600

tanisxu, 2018-01-15:
@xie-dongping good point, thank you.

closed 2018-03-05 by TheNihilisticRobot

kargarisaac, 2019-09-20:
Hi,
Thank you guys for this information. Which one do you suggest to start with? I want to use it for multi-agent reinforcement learning. I want to build some scenarios in different conditions. I also want to use Autoware. I saw a github repo for carla-autoware. 
In general which one do you suggest?

Thank you

Leviathan321, 2024-08-24:
Are there any new updates regarding the comparison of Carla vs. Airsim. @kargarisaac Did you could integrate autoware with Airsim? I am looking for comparable simulators such as Carla to integrate some SOTA system under test such as Autoware

zhefang-thu, 2024-11-29:
> 关于 Carla 和 Airsim 的比较有任何新的更新吗？@kargarisaac你能将 autoware 与 Airsim 集成吗？我正在寻找类似的模拟器，例如 Carla，以集成一些正在测试的 SOTA 系统，例如 Autoware

@Leviathan321 I noticed that a project had been completed, namely UAV3D [https://github.com/huiyegit/UAV3D](url). However, they did not provide details about the data collection process.