carla-simulator/carla #114: Server not working

CapitalZe, 2018-01-05:
Hi,

I repeatedly get the error:
LogCarlaServer: Warning: Client Disconnected, server needs restart

This only happens if I enter the command line option 'carla-server'

If I just run ./CarlaUE4.sh with any other command line option then it works fine, but this is not what I am aiming for.

Does anyone have any idea what's going on?

felipecode, 2018-01-05:
Hey @CapitalZe 
It seems that you are not connecting the python client.
On this mode the server waits for the python client to control the car.
Do like this
Run the server in carla server mode
$ ./CarlaUE4.sh /Game/Maps/Town02 -carla-server
Then run the python client:
$ ./client_example.py

CapitalZe, 2018-01-05:
Hi Felipe,

Thanks so much for your reply. I really appreciate it. I actually spent yesterday reading your paper and and have since been trying to run your CIL.py file.

I have just tried what you said and coming up with the following:

Traceback (most recent call last):
  File "/home/nicholas/Desktop/PythonClient/carla/client.py", line 19, in <module>
    from . import carla_server_pb2 as carla_protocol
  File "/home/nicholas/Desktop/PythonClient/carla/carla_server_pb2.py", line 6, in <module>
    from google.protobuf import descriptor as _descriptor
ModuleNotFoundError: No module named 'google'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "PythonClient/client_example.py", line 19, in <module>
    from carla.client import make_carla_client
  File "/home/nicholas/Desktop/PythonClient/carla/client.py", line 21, in <module>
    raise RuntimeError('cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file')
RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file

I can't figure our why 'carla_server_pb2.py will not import. I also am struggling to see why the google module is not found, I have tried redirecting it to where it is on my system.

Really, your input is very much appreciated and fascinating paper.

CapitalZe, 2018-01-05:
*python run_CIL.py

felipecode, 2018-01-05:
It seems that you are running the example from the other repository,
the imitation learning code.
Also, It seems that you don't have the dependencies.
Please check this link http://carla.readthedocs.io/en/latest/how_to_run/
to install. You are missing the protobuf. Try to install it as well for python2
since the imitation learning is not yet ready for python3.

mvpcom, 2018-01-05:
After @felipecode suggestions, I recommend you to make sure [this ](https://github.com/carla-simulator/carla/commit/30c54019ec84d533d0d3d5082b277c2c93040cbb) is not your problem. Because that can cause this error too: "RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file"

felipecode, 2018-01-07:
issue closed please open a new one if the problem persist!

closed 2018-01-07 by felipecode

CapitalZe, 2018-01-08:
Thanks everyone. I will try the recommendations today.

CapitalZe, 2018-01-08:
Update for future reference:

Both pieces of advice resolved the problem for getting the client_example.py to connect to the carla server. Thank you.

AliBaheri, 2018-10-17:
I still have the following problem:

` raise RuntimeError('cannot import "carla_server_pb2.py", run '
RuntimeError: cannot import "carla_server_pb2.py", run the protobuf compiler to generate this file`

even after installing "protobuf" on Ubuntu 18.

@CapitalZe  How did you resolve this? Thanks.

aroongta, 2018-10-23:
I too am facing the same problem as @AliBaheri .
Can anybody please help me resolve this?
Thank you

raozhongyu, 2020-11-24:
I meet the same question, do you solve it?

yajat23100, 2023-06-01:
my cmd is not working