carla-simulator/carla #118: A question about CarlaWheeledVehicle

tanisxu, 2018-01-08:
I've notice in 
https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/CarlaWheeledVehicle.cpp
there's a function named GetVehicleMovementComponent()
but i can't find it anywhere. Is it in a generated file or something?
Thank you!

nsubiron, 2018-01-08:
Hi @tanisxu, it's a function of the base class [AWheeledVehicle](https://docs.unrealengine.com/latest/INT/API/Plugins/PhysXVehicles/AWheeledVehicle/).

closed 2018-01-09 by nsubiron

sibojia, 2018-01-10:
@nsubiron I noticed that AWheeledVehicle is a vehicle dynamic model provided by NVIDIA PhysX. The vehicle parameters it supports are not very well documented. In autonomous driving industry CarSim may be a more professional choice. Have you compared these two libraries, and is there a plan to support external vehicle models?

nsubiron, 2018-01-10:
Hi @sibojia, please check out issue #115.