carla-simulator/carla #120: Windows binaries

eds89, 2018-01-08:
Hello. Please can you make windows binaries available? I'm having difficulty in compiling it.

nsubiron, 2018-01-08:
Hi @eds89, although we plan to add it in the future, we currently do not support Windows as we too are having difficulties compiling it ;)

eds89, 2018-01-08:
Hi,
Thank you so much for your reply.

May I ask if I do manage to compile it on Windows, will I be able to run
Carla on it? Or, is it only the plugin that would run? To be honest, I
didn't understand what the plugin does in unreal. Would it add the cars to
any other unreal game? :-)

I'm planning to extend Carla as part of my PhD project. I wonder whether
you have implemented any sensor fusion algorithm on the player car?
Moreover, am I correct to assume that, to add new post-processing effects,
I need to extend the "SceneCaptureCamera.cpp" file?

nsubiron, 2018-01-09:
Hi @eds89,

Yes, it is possible to run CARLA on Windows, however it needs some extra work as we don't have an automated system yet. Please follow the discussion at #21.

The plugin contains most of the functionality of CARLA, without any of the assets (3D models, etc.). In principle, it can be added to any project and will provide the tools and base classes to build CARLA, say you wanted to build your own CARLA with your own assets. You wouldn't have any of the vehicles, but the base class with the controllers, autopilot and so on. In practice however, this would be a lot of work as there are many pieces to fit and we are still working on documenting everything. We recommend to get the whole project and modify it for your needs.

CARLA does not provide any data fusion algorithm, it just provides the API for you to retrieve the data. This is how it works, the CARLA server runs the simulation, renders the images, and produces all the measurements. We provide a Python API to communicate with the server: set up the simulation, control the car, and retrieve the data; but it's up to the user to code the actual functionality of the client. We have a minimal example of a client using the API at [PythonClient/client_example.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py).

And yes, if you wanted to add another post-processing effect you need to extend the SceneCaptureCamera. You create a new material as the ones in "Unreal/CarlaUE4/Plugins/Carla/Content/PostProcessingMaterials/", then load it at "SceneCaptureCamera.cpp" and apply it to the camera, just look at how the depth and semantic segmentation are done.

Cheers!

closed 2018-01-15 by nsubiron