carla-simulator/carla #121: ModuleNotFoundError no module named 'carla'

CapitalZe, 2018-01-09:
I can run the manual and example clients on the carla-server with no problem.

When I try to run 'python run_CIL.py' I am returned with this:

:~/Desktop/Codevilla/imitation-learning$ python run_CIL.py
Traceback (most recent call last):
  File "run_CIL.py", line 5, in <module>
    from carla.PythonClient.benchmarks.corl_2017 import CoRL2017
ModuleNotFoundError: No module named 'carla'

With sudo it kicks up this:

~/Desktop/Codevilla/imitation-learning$ sudo python run_CIL.py
Traceback (most recent call last):
  File "run_CIL.py", line 5, in <module>
    from carla.PythonClient.benchmarks.corl_2017 import CoRL2017
ImportError: No module named carla.PythonClient.benchmarks.corl_2017

The modules are there in the pointed directory.

Any ideas?

Also posted in < https://github.com/carla-simulator/imitation-learning/issues/2 >

Thanks.

CapitalZe, 2018-01-09:
I overcame this by placing all pertinent modules in anaconda sitepackeges, but then I still am unable to get run_CIL. working.
The next problem that arises is:

File "/home/nicholas/anaconda3/lib/python3.6/site-packages/carla/PythonClient/benchmarks/experiment.py", line 2, in <module>
    from carla.settings import CarlaSettings
ImportError: cannot import name 'CarlaSettings'

I've gone through every step meticulously. Trying every variant.

Is there perhaps anyway to that through carla locally?

My goal is to get the CIL agent/code running on Carla, I'm really struggling.

Cheers!

YashBansod, 2018-03-13:
@CapitalZe can you just try and copy the carla folder (the one in the PythonClient folder) to the folder your code is in. I think that should solve the problem.

felipecode, 2018-03-20:
I think it was already answered on https://github.com/carla-simulator/imitation-learning/issues/2.  Closing ! If any more help needed please open again, or open another issue.

closed 2018-03-20 by felipecode