carla-simulator/carla #122: Is there a way to generate a map better than just drag spline repeater?

tanisxu, 2018-01-09:
I came across most parts of code and learn that for now, map generator only supports 90 degree straight road generation, but there's another way to generate roads that can be in any angle and curly, which is **spline mesh repeater**, I see in doc. But using this method seems very cumbersome, we have to drag point in ue editor. And there's a collision issue in this **repeater**,
Is there a better way such as use a script to generate map? Or will there be a resolution recently?
Thank you!

nsubiron, 2018-01-09:
Hi @tanisxu,

We only support 90 degree straight roads for now. Keep in mind that if you generate a road with the spline repeater the lane information will be missing, meaning the vehicle will appear always as driving off-road and the autopilot won't work. However, if you still want to use it you can easily script it with Unreal blueprints.

We will soon start working on a better map generator with curvy roads and a varying number of lanes, but it will take time as it is a fairly complex addition.

tanisxu, 2018-01-10:
@nsubiron Thanks for the warm and quick answering all the time. May I ask how long will the curly(and maybe, with height) map generator takes? And will you add height to map?
Thanks again.

nsubiron, 2018-01-10:
@tanisxu Yes, we plan to have maps with changes in height. We plan to work on this along this coming year, but I cannot be more specific with deadlines since we haven't planned yet how we will do it and how many features we will add.

tanisxu, 2018-01-11:
@nsubiron Thank you for your answer. May I ask how long will the curly road map generator takes?

nsubiron, 2018-01-11:
@tanisxu Sorry, we don't know yet. We'll be more specific once we start working on it.

tanisxu, 2018-01-11:
@nsubiron Thanks all the same.

closed 2018-01-18 by nsubiron