carla-simulator/carla #123: What does 'face' mean in the stage of map generation?

besttangent, 2018-01-09:
https://github.com/carla-simulator/carla/blob/9855fe35f9691a4e9346f6a4525de08777c9cbf9/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/DoublyConnectedEdgeList.h#L32
Does it mean a direction of the half-road?Like north,southand so on.
Thanks.

nsubiron, 2018-01-09:
Hi @besttangent, 

A face is part of the [DCEL](https://en.wikipedia.org/wiki/Doubly_connected_edge_list) data structure, represent the closed area in between edges. It is not currently holding any useful information apart from helping to iterate the DCEL. In our map, it would be the area in between roads, e.g. a block with buildings.

besttangent, 2018-01-10:
Ok,I understand it.Thank you.

closed 2018-01-10 by besttangent