carla-simulator/carla #125: Loading assets

eds89, 2018-01-10:
Hello there.

How can I load a new camera postprocessing effect (uasset) without having to re-compile Carla? Is that possible at this stage?

nsubiron, 2018-01-10:
As it is now you need to modify the C++ code to load a new post-processing effect, so you need to recompile.

eds89, 2018-01-26:
Hi. Where, in the C++ code, do I add new Post Processing Effects?

nsubiron, 2018-01-26:
Hi, 

If you mean a post-processing effect like "depth" or "semantic segmentation", you need to create a new material (shader) inside Unreal editor. Then add it to the SceneCaptureCamera C++ class as a UMaterial member, just follow the way it's done for `PostProcessSemanticSegmentation` for instance.

If on the other hand, what you want is to change other visual post-processing effects, like flares, fog, etc. Then you need to change them in the "PostProcessVolume" present in the scene, search for it in the "World outliner" tab. In this case there is no need to change the C++ code.

eds89, 2018-01-26:
Hi,
It's a camera effect like the Semantic Segmentation one. Is there any
practical way to visualise how the effect looks like in Carla without
having to compile from source every time? I couldn't understand the
Semantic Segmentation effect when I opened it on Unreal Editor. UE uses
some sort of visual scripting language, unfamiliar to me.

nsubiron, 2018-01-26:
You can add the material to a post-process volume, if you enter the volume while navigating the 3D scene the post-process material will apply to your view, even in editor mode. Just remember to remove it once you are done.

TheNihilisticRobot, 2018-02-01:
I'll reopen this Issue to clarify we are adding this to the documentation.

germanros1987, 2020-03-19:
This is now totally possible using CARLA API, so I am closing this issue.

closed 2020-03-19 by germanros1987

madhu1sept, 2022-12-16:
Hi,
Currently I am adding new HDR(float camera) in Carla. 
I need to add new postprocessing in this HDR camera.
Do adding  AddPostProcessingMaterial(
      TEXT("Material'/Carla/PostProcessingMaterials/Imox'")); in  HDR class (in plugin folder) will apply this on  HDR Camera?