carla-simulator/carla #127: Get exact pixel location from town02 and the town02Map.png file. What is the scaling factor.

LoayMAmin, 2018-01-11:
I have successfully mapped town01 from the simulator position to the map (.png) image you included.
This is done with certain scaling factor (1643 to be exact) in town01

What is the number for town02?
1643 does not seem to work for me.

nsubiron, 2018-01-11:
Hi @LoayMAmin, I think that's because the Town02 road layout is not placed at `(0.0, 0.0, 0.0)` but somewhere else, at `(544.0, -10748.0, -22.0)`. Yup it wasn't the best choice...

I have open #129 because indeed this is missing documentation. Please check out the files attached, there you have all the info necessary to convert pixels to world coordinates.

closed 2018-01-11 by nsubiron

LoayMAmin, 2018-01-11:
Hi @nsubiron, thank you for the fast reply! But I'm still confused I'm sorry. I'll tell what I'm doing on town01 tat works perfectly but can not do on town02.

Town01:

> I get the measurments from read_data()

`measurements, sensor_data = carlaa.read_data()`

> Then I get the location of the car in the simulation

` x = measurements.player_measurements.transform.location.x
  y = measurements.player_measurements.transform.location.y`

> Then I translate these measurments to get the coressponding pixel in the town01.png  image

` x_pixel = (x + **1643**) // 16.43
  y_pixel = (y + **1643**) // 16.43`

This method works well on town01. But fails to work on town02. I believe this is due to this number **1643** which is the scaling factor from the simulator to the map.

So what is the required number to use in town02?

nsubiron, 2018-01-11:
From the txt file you get the necessary parameters for the transformation

```
Map name = Town02
Size = 1,400x1,400 pixels
Density = 16.43 cm/pixel
World-To-Map Transform (T|R|S) = (544.000000,-10748.000000,-22.000000|0.000000,0.000000,0.000000|1.000000,1.000000,1.000000)
Map Offset = (X=-1643.022 Y=-1643.022 Z=0.000)
```

In Python

```py
world_to_map_translation = numpy.array([544.0, -10748.0])
map_offset = numpy.array([-1643.022, -1643.022])
cm_per_pixel = 16.43
```

You can ignore the rotation and scale.

Then convert it following the formulas shown in the pdf doc

```py
measurements, sensor_data = client.read_data()

world_location = numpy.array([
    measurements.player_measurements.transform.location.x,
    measurements.player_measurements.transform.location.y])

relative_location = world_location + world_to_map_translation - map_offset

pixel_coord = relative_location // cm_per_pixel
```