carla-simulator/carla #130: Altering uassets

eds89, 2018-01-11:
Hi. I'm new to Unreal Engine. What would the easiest way to extend the current camera post-processing materials for Carla?
All the best.

nsubiron, 2018-01-11:
Hi @eds89. To modify an existing uasset, just open the project with Unreal Editor. Look for the uasset in the content browser and double-click it to open. Modify it and save it.

If you need to add a new post-processing material to the camera you will need to modify the code in "SceneCaptureCamera.cpp". Just look in the code how the current post-processing materials are used and do the same with the new one.

closed 2018-01-11 by nsubiron

eds89, 2018-02-16:
Hi,
As advised, I've changed the file. Compilation went without errors. However, after opening the compiled project on Unreal Engine, I cannot find the new added assets in the content pane. Is that normal?

Besides altering SceneCaptureCamera.cpp, I also altered Settings.cpp (I think) to enable settings the new camera effect via settings file.

nsubiron, 2018-02-16:
Hi @eds89,

Yes, the plugin content is hidden by default, check out https://github.com/carla-simulator/carla/issues/180#issuecomment-361199548 for instruction to show them in the content browser.