carla-simulator/carla #131: Running ./Setup.sh

RohanSaphal, 2018-01-11:
Hi,

I am not able to get the required results while running ./Setup.sh. 

I am getting the following error when i run Setup.sh.

rohan@rohan-Lenovo-Y520-15IKBN:~/carla-master$ ./Setup.sh
Downloading Content.tar.gz...    32.77KB                                                             Done.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

Can anyone tell me whats wrong?

nsubiron, 2018-01-12:
Hi @RohanSaphal, something went wrong with the download step, the Content.tar.gz file is about 10GB and you only downloaded few KB. Try running it again, if not you can download it manually from [here](https://drive.google.com/open?id=11PKC_lG7L80L1ZxiMlzV17utU45UcRnT).

RohanSaphal, 2018-01-12:
Thanks @nsubiron 

How do i build the dependencies after that?..and where to unzip the file.

nsubiron, 2018-01-12:
The content should be extracted at `Unreal/CarlaUE4/Content`.

Now, I just realized we don't have a nice way to skip the content download in the Setup script. You'll have to use a small hack, just create an empty folder called `.git` inside the newly create Content folder and run the setup script again, it will skip the download and compile the dependencies.

    $ mkdir Unreal/CarlaUE4/Content/.git
    $ ./Setup.sh

RohanSaphal, 2018-01-13:
I am getting this error when using the rebuild script...

Can u tell me whats going wrong?

Performing 11 actions (8 in parallel)
[2/11] Compile Module.Carla.cpp
[1/11] Compile PCH.CarlaUE4.h
[3/11] Compile Module.Carla.gen.cpp
fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header
      '/home/rohan/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/rohan/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header
      '/home/rohan/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/rohan/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
1 error generated.
1 error generated.
[4/11] Compile CarlaUE4GameModeBase.gen.cpp
[5/11] Compile CarlaUE4.init.gen.cpp
[6/11] Compile CarlaMapGenerator.gen.cpp
[7/11] Compile CarlaUE4GameModeBase.cpp
[8/11] Compile CarlaUE4.cpp
[9/11] Compile CarlaMapGenerator.cpp
ERROR: UBT ERROR: Failed to produce item: /home/rohan/carla-master/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
Total build time: 149.76 seconds (Local executor: 0.00 seconds)
Makefile:600: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 5

nsubiron, 2018-01-15:
Yes, this is a common issue, check out our [Troubleshooting](http://carla.readthedocs.io/en/latest/troubleshooting/) section. You should force a recompile and possibly reboot the computer
```
$ cd Unreal/CarlaUE4/
$ make CarlaUE4Editor ARGS=-clean
$ make CarlaUE4Editor
```

closed 2018-01-22 by nsubiron

WillenPeng, 2018-08-15:
Hi,

Could you provide the download link for Content_0.8.2 instead of Content_0.7.0?

nsubiron, 2018-08-20:
Hi @WillenPeng , checkout #652.