carla-simulator/carla #133: Instant Collision Intensity

mvpcom, 2018-01-12:
I don't want the cumulative collision intensities. Is there any way to receive the instant collision intensities for each frame?

mvpcom, 2018-01-12:
I found in CarlaPlayerState.cpp the following code:
```
switch (ATagger::GetTagOfTaggedComponent(*Hit.Component)) {
    case ECityObjectLabel::Vehicles:
      CollisionIntensityCars += NormalImpulse.Size();
      break;
    case ECityObjectLabel::Pedestrians:
      CollisionIntensityPedestrians += NormalImpulse.Size();
      break;
    default:
      CollisionIntensityOther += NormalImpulse.Size();
      break;
  }
```

Please don't tell me the only way for accessing to instant collision intensity in the simulator is changing the above code and recompiling the whole simulator? Although one way is using some tricks like have another variable which keeps previous value and then calculate the difference in both variables and find out the instant collision for that specific image, I think it would be great if there is some configurable variable and receive whatever the client wants. If there is not exist any easy way, I assume this will be a feature request.

nsubiron, 2018-01-12:
Hi @mvpcom, you could just subtract the collision intensity of the previous frame.

mvpcom, 2018-01-13:
Another question is about the impact of collision measurements. What the meaning of impacts? How do you measure and report the collision values in the simulator? Please let me know more about the definition of impact and unit of cumulative collisions.

nsubiron, 2018-01-15:
The collision intensity is the norm of the normal impulse between the two colliding objects as reported by the physics engine (expressed in kg*cm/s).

Please check out our [Measurements](http://carla.readthedocs.io/en/latest/measurements/) documentation.

closed 2018-01-22 by nsubiron

AOOOOOA, 2023-02-20:
> The collision intensity is the norm of the normal impulse between the two colliding objects as reported by the physics engine (expressed in kg*cm/s).
> 
> Please check out our [Measurements](http://carla.readthedocs.io/en/latest/measurements/) documentation.

Hi I still don't get the answer to the question proposed by @mvpcom after reading the measurement document. How does Carla determine the collision happen? I only found the momentum of the intensity in the code without an explanation of the collision determine criterion. Could you please give some suggestions? @nsubiron Thanks so much.