carla-simulator/carla #137: Running Carla on the Cloud

eds89, 2018-01-15:
Hello,
Does anyone have any experience in running CARLA on the cloud? It seems as the only available alternative to get it running to finalise my research. My computer has got only an onboard graphics card.

dosovits, 2018-01-25:
Hi @eds89 , sorry about the delay! We have run CARLA on AWS. It's not trivial, but doable.

Here is brief info on the docs:
https://github.com/carla-simulator/carla/blob/master/Docs/how_to_run.md#running-carla-off-screen

Here is a more detailed tutorial (still work in progress by @felipecode ):
https://github.com/carla-simulator/carla/blob/tutorial_for_running_headless/Docs/carla_on_server.md

@felipecode , @nsubiron Perhaps one day we should provide some pre-packed AWS installation for people to use.

eds89, 2018-01-25:
> Docs/how_to_run.md#running-carla-off-screen
> for_running_headless/Docs/carla_on_server.md

Hi,
Can you confirm that I need to follow these instructions even if I'm trying
to run Carla on a GUI-enabled virtual machine? I guess it must be a problem
with incorrect graphics card selection, but given that I'm not an expert,
it can be anything really. The machine has got plenty of resources: 12
cores, 56 GB RAM and 16 GB of GPU memory (NVidia Tesla). It might be a
problem with the graphics setup, as the OS (Ubuntu 16.04) doesn't recognise
it.

I'll try your suggestions though.

dosovits, 2018-01-25:
Hi,
It is difficult to say for sure: these display-related settings are sometimes quite complicated, and we observed different behavior on different systems. I think the best approach is for you to try, use these materials I linked above as guidance, and write here again if you encounter problems, or also if you succeed. Maybe with GUI it will just work out of the box. If you only have 1 GPU, you shouldn't need turbovnc and virtualgl.

Lucaccio1, 2018-02-01:
Hi @eds89 did you solve your problem in the last week? I'm having your same issue probably.

Me and my fellows are using a Microsoft Azure VM (since my university is having a test of this product) that has, pretty much, your same configuration. 
I'm facing a signal 11 issure that I found online that it's related to GPU and graphic drivers even when I started just UE4 Editor. B
elow there's an image of my video cards configuration. 

Thanks

OT: I was able to let CARLA run on my pc (I hade the same issue switching through video cards but I solved it with NVIDIA X SERVER SETTING) but since it's a DELL PRECISION T5500 it was too slow (3 FPS)
[image]

wendersonj, 2019-02-13:
> Hi @eds89 , sorry about the delay! We have run CARLA on AWS. It's not trivial, but doable.
>
> Here is brief info on the docs: [...]

Hello ! An pre-packed AWS would be awesome !! I'm having real headache in running Carla on AWS and I'm following the tutorial for headless run, but having some trouble on the finals commands with `Xorg`.

knmitri, 2019-08-23:
Until today, the GPU drivers provided on the cloud do not support OpenGL 4.3. That is one main issue.

eds89, 2019-09-14:
More urgently is when will there be a back to back tutorial on how to use
Carla?

wendersonj, 2019-09-18:
> More urgently is when will there be a back to back tutorial on how to use Carla?
> […](#)

Hi ! I want to record some videos teaching how to use Carla and post it on YouTube. Probably early October. Cheers!

soheilade, 2020-02-12:
any body managed to run CARLA on a cloud VM?

knmitri, 2020-02-12:
> any body managed to run CARLA on a cloud VM?

1) The machine should have a GPU  
2) In case of Ubuntu, connect using VNC
3) Install the needed librairies
4) Innovate :)

Cheers 🥂

germanros1987, 2020-03-19:
This is now easy to do by following the instructions here: https://carla.readthedocs.io/en/latest/build_docker/

closed 2020-03-19 by germanros1987

thefatbandit, 2020-06-23:
Thanks Jbnunn https://github.com/jbnunn/CARLADesktop.
Succesfully run carla on AWS EC2.