carla-simulator/carla #140: Any suggestion on understanding carla structure?

tanisxu, 2018-01-16:
Hello all, I'm currently learning UE4 so that I can understand carla source code clearly. But I found it hard to find the right tutorial, there're so many course about blueprint and so on.
What roadmap of learning will you guys prefer for one to understand carla source and develop new features for community?

nsubiron, 2018-01-16:
Hi @tanisxu,

We are writing basic notions of CARLA design [here](http://carla.readthedocs.io/en/latest/carla_design/), though it is still super minimal.

I'm assuming you have some notions of C++, as UE4 code is written in this language you would need at least basic notions of object oriented programming with C++.

A basic introduction to C++ programming with UE4 can be found at Unreal's [C++ Programming Tutorials](https://docs.unrealengine.com/latest/INT/Programming/Tutorials/). For working with Blueprints, as you said, there are many tutorials online, any of them should do.

Then, if you want to dive into UE4 C++ development, the [Unreal Course at Udemy](https://www.udemy.com/unrealcourse/) it's pretty good and there are usually offers that make it very affordable, with section 3 you can probably get enough idea to start understanding CARLA code.

Then you can take a look at our own classes, for an example of how C++ classes work on UE4 you can take a look for instance at ["ASceneCaptureToDiskCamera"](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/SceneCaptureToDiskCamera.h) (and its parent class "ASceneCaptureCamera"). This class creates an actor that can be dropped into the scene. In the editor, type "Scene Capture To Disk" in the Modes tab, and drag and drop the actor into the scene. Now searching for its detail tab you can find all the "UPROPERTY" members reflected. This shows the basic mechanism to use C++ classes in Unreal Editor.

For a more advanced example on how to extend C++ classes with blueprints, you can take a look at the "VehicleSpawner" blueprint. It derives from the C++ class "AVehicleSpawnerBase". The C++ class decides where and when it spawns a vehicle, then calls the function "SpawnVehicle", which is implemented in the blueprint. The blueprint then decides model and color of the vehicle being spawned. Note that the blueprint can call back C++ functions, for instance for getting the random engine. This way there is a back-and-forth communication between C++ code and blueprints.

Hope I'm giving you something to start with :)

tanisxu, 2018-01-16:
@nsubiron Great! Very helpful! Thank you!

commit eb9403b4 (2018-02-02): Close #140 tips for learning Unreal
closed 2018-02-03 by nsubiron