carla-simulator/carla #142: Improve documentation

nsubiron, 2018-01-16:
Our documentation needs a boost!

- [ ] Hands-on tutorial
- [ ] How to run headless
- [ ] How to create your own scenarios with our tools
- [ ] More FAQ/Troubleshooting (get from [questions list](https://github.com/carla-simulator/carla/issues?q=is%3Aissue+label%3Aquestion+sort%3Acomments-desc))
- [ ] Suggestions?

All contributions are appreciated ;) 

Submit your commits/pull-requests to the `documentation` branch.

felipecode, 2018-01-18:
Adding a clear and aesthetically pleasing project roadmap, both in short and in long term goals.

germanros1987, 2020-03-19:
New documentation can be found here: https://carla.readthedocs.io/en/latest

Thanks to @sergi-e for his amazing work!

closed 2020-03-19 by germanros1987