carla-simulator/carla #143: Logging data in a "build from source"

TrygveG, 2018-01-16:
I was wondering if it is possible to log data from the unreal editor built from source, and if so, if there are any examples on how to run that?

nsubiron, 2018-01-17:
Yes, you can connect the Python client to a server running within Unreal Editor as if it was the standalone server (slightly slower though).

Go to **"Unreal/CarlaUE4/Config/CarlaSettings.ini"**, this file should have been created by the Setup.sh, and enable networking. If for whatever reason you don't have this file just create it and add the following
```ini
[CARLA/Server]
; If set to false, a mock controller will be used instead of waiting for a real
; client to connect.
UseNetworking=true
```
Now when you hit Play the editor will hang until a client gets connected.

TrygveG, 2018-01-17:
Thanks for the quick reply. I have tried with both building from source and running the python client, but I am having issues with connecting to the client in both cases. Is there any way I need to initialize the connection? 
In the build from source, I have ran the "Launch the editor" script, and then started a simulation (which works without using "UseNetworking = true").
I have also tried the instructions from "Running the server".
I am attaching the console output from both cases. 
[Local build](https://github.com/carla-simulator/carla/files/1639562/serverIssues.docx)
[CARLA server and client](https://github.com/carla-simulator/carla/files/1639563/serverIssuesNonLocal.docx)

nsubiron, 2018-01-17:
Your server seems to run fine, it should hang at the point

    LogCarlaServer: Waiting for the client to connect...

Make sure that the `WorldPort` in your CarlaSettings.ini matches the port you are giving to the client (by default port 2000), and that your firewall is allowing both applications to connect.

If you are running them in different machines/port, launch the client with

    $ ./client_example.py --port=SERVER_PORT --host=SERVER_IP -v

Use too `-v` to print debug output.

closed 2018-02-01 by carla-admin