carla-simulator/carla #144: Camera Reading

sherifdarwish92, 2018-01-17:
I am trying to get the camera reading by printing object of Image and Camera0 (which is in the client example) but it always print a value which is cameradepth:<carla.sensor.cameraobject at 0x7f1823224550>
So I want to now how to get the value of camera (BGRA Array)
Thank you very much.

n-akram, 2018-01-17:
Hi @sherifdarwish92 ,

If you want to save camera image as jpeg or png files, you can use default arrays which are sent to "self.img_vec" in 'carla_manual_control.py'. Sample code snippet to save images as jpeg, is as follows. It uses, PIL library.

im = Image.fromarray(self.labels_vec[0])
background = Image.new("RGB", im.size, (255, 255, 255))
background.paste(im, mask=im.split()[3]) # 3 is the alpha channel
background.save(name)

Hope this helps.

nsubiron, 2018-01-17:
Hi @sherifdarwish92,

This is how you do it with the 0.7.0 API and later.

The `sensor.Camera` object defines a new sensor that is going to be added to the car, you need to add it to the settings object (please check out our [client_example.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py) for a whole example)

```py
settings = CarlaSettings()
my_camera = Camera('MyCamera', PostProcessing='Depth')
settings.add_sensor(my_camera)
# ...
carla_client.load_settings(settings)
```
For each camera you add, you retrieve a `sensor.Image` object every frame

```py
measurements, sensor_data = carla_client.read_data()
image = sensor_data['MyCamera']
```

Now from that image object you can be extract the data in different formats

```py
# Image as a buffer of uint8s.
image.raw_data

# Default format (depends on the camera PostProcessing but always a numpy array).
image.data

# numpy BGRA array.
image_converter.to_bgra_array(image)

# numpy RGB array.
image_converter.to_rgb_array(image)
```

If you wish to save the image as PNG

```py
image.save_to_disk('image.png')
```

sherifdarwish92, 2018-01-18:
Thank you very much for your help your info is really helpful and
supportive.

Thank you again.

sherifdarwish92, 2018-01-22:
Hello,
I have printed the camera reading but when I printed It wasn't very descriptive as when I printed the camera while making the mode Semantic I found the output as in this photo
[image]
However I expected the data to contain the number from 0:12 as described in the docs that objects varies from number 0:12 in the semantic mode.
and when I get the data by the mode of Scene final I found the output as in this pictures
[image]
[image]
which was very ambiguous for me as I need to transfer the data from the camera 
even when I print the size of the BGRA array the output is 600
so could you please provide me with the description of the output of the camera in the three modes( semantic - depth - scene final) in order to be able to send the value of the output of the camera.
Thank you very much for your help.

nsubiron, 2018-01-22:
The output is weird indeed. This is what you are supposed to get:

Using the built-in "save_to_disk" function
```py
image.save_to_disk('my_image.png')
```
**SceneFinal**
[image]
**Depth**
[image]
**SemanticSegmentation**
_Dark tones of red._
[image]

and printing the RGB array
```py
print(image_converter.to_rgb_array(image))
```
**SceneFinal**
```
[[[ 66  98 137]
  [ 66  98 137]
  [ 66  96 137]
  ..., 
  [ 73  98 133]
  [ 73  98 133]
  [ 71  95 129]]

 [[ 69  98 135]
  [ 69  98 136]
  [ 69  98 137]
  ..., 
  [ 73  98 132]
  [ 73  98 131]
  [ 73  97 129]]
```
**Depth**
```
[[255 255 255]
  [255 255 255]
  [255 255 255]
  ..., 
  [255 255 255]
  [255 255 255]
  [255 255 255]]

 ..., 
 [[ 70  38   0]
  [ 70  38   0]
  [ 70  38   0]
  ..., 
  [ 70  38   0]
  [ 70  38   0]
  [ 70  38   0]]
```
**SemanticSegmentation**
```
 [[ 0  0  0]
  [ 0  0  0]
  [ 0  0  0]
  ..., 
  [ 0  0  0]
  [ 0  0  0]
  [ 0  0  0]]

 ..., 
 [[10  0  0]
  [10  0  0]
  [10  0  0]
  ..., 
  [10  0  0]
  [10  0  0]
  [10  0  0]]
```

commit 6ce87668 (2018-02-02): Close #144 add sample output images to docs
closed 2018-02-03 by nsubiron