carla-simulator/carla #146: Error while downloading CARLA

ajax98, 2018-01-17:
Hi,

I am trying to build Carla for linux and  was running the Rebuild.sh script where I got two errors: fatal error: 'carla/carla_server.h' file not found
#include <carla/carla_server.h> and ERROR: UBT ERROR: Failed to produce item: /home/ajay/carla-master/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so.  Do you know why these errors might be occuring?

nsubiron, 2018-01-17:
It seems like the CarlaServer dependency failed to compile.

Make sure that the Setup script does print _"Success!"_ at the end

    $ ./Setup.sh
    [...]
    ****************
    *** Success! ***
    ****************

Then check if CarlaServer compiles without errors running make

    $ make

It should end printing something like

```
[1/1] Install the project...
-- Install configuration: "Release"
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++abi.so.1
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++abi.so.1.0
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++.so.1
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++.so.1.0
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++.so
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/shared/libc++abi.so
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libc++abi.a
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libboost_system.a
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libprotobuf.a
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/include/carla
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/include/carla/carla_server.h
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libcarlaserver.a
-- Installing: Unreal/CarlaUE4/Plugins/Carla/CarlaServer/bin/test_carlaserver
-- Set runtime path of "Unreal/CarlaUE4/Plugins/Carla/CarlaServer/bin/test_carlaserver" to ""
```

If so you can safely run Rebuild.sh.

ajax98, 2018-01-18:
Thanks, I was able to fix this.

muzaheed57, 2018-01-23:
i am in windows. having the same issue.  running Setup.sh is not doing anything

nsubiron, 2018-01-23:
Hi @muzaheed57,

We don't have full support for Windows yet, you will need to compile the external dependencies manually if you want to run CARLA in Windows. Please check out #21.

commit 1b55e1ff (2018-02-01): Close #146, add common error to faq
closed 2018-02-03 by nsubiron