carla-simulator/carla #147: OpenGL 4.3 not supported by driver, how can i handle it ?

SoonHoJang, 2018-01-18:
When I follow the calar documentation, I get the following problem at code(rebuild part)

```bash

$ UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh

```

it says 

```bash

[2018.01.18-01.31.46:238][  0]LogInit: Error: _PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2018.01.18-01.31.46:238][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
[2018.01.18-01.31.46:239][  0]LogInit: Error: _PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'

```

like this

nsubiron, 2018-01-18:
Hi @SoonHoJang,

It might be a matter of installing the latest nvidia drivers, see https://github.com/carla-simulator/carla/issues/73#issuecomment-353768715.

If it still doesn't work after this, you can try launching the editor with OpenGl 3

```
$ cd Unreal/CarlaUE4
$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor -opengl3 "$PWD/CarlaUE4.uproject"
```

closed 2018-02-01 by carla-admin

J-zin, 2019-03-30:
@snowjang24  how to fix it, i got the same problem

ahnsangjae, 2019-06-26:
i got the same problem too, how to fix it? plz let me know~!

473867143, 2019-08-10:
i got the same problem too, finally,i solved it.
i'm sure to be install MVIDIA drivers,but it doesn't correct display.
      cd ~/UnrealEngine/Engine/Binaries/Linux
      ./UE4Editor -opengl3 "$PWD/CarlaUE4.uproject"

furqan5411, 2020-01-25:
This error
[2020.01.25-13.40.45:326][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2020.01.25-13.40.45:326][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
[2020.01.25-13.40.45:331][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'