carla-simulator/carla #150: Building Carla on Mac OSX

ajax98, 2018-01-19:
Online I see that there only exist manual for installing and building Carla on Linux. Is it possible for installing and running it on Mac?

nsubiron, 2018-01-19:
Hi @ajax98,

Yes, it is possible. However we don't have an automated build system for MacOS, so it would require some effort to make it work.

Please have a look at the progress at #103. Also it might be helpful to check out the documentation for [How to build on Linux](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/) and the discussion for building on Windows at #21 as the process is basically the same.

Please share your progress here if you find any difficulties. It can be very helpful for other people trying to build on Mac, and eventually we might be able to make an automated system as for Linux.

- - -

There is a compilation error that is going to appear, you need to replace the block at "Source/Carla/SceneCaptureCamera.cpp" Line 27 by
```cpp
static constexpr auto DEPTH_MAT_PATH =
#if PLATFORM_LINUX
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial_GLSL.DepthEffectMaterial_GLSL'");
#else
    TEXT("Material'/Carla/PostProcessingMaterials/DepthEffectMaterial.DepthEffectMaterial'");
#endif
```
This is something we have to fix, but we'll have to test if the default depth material is working well on Mac. Help on this would be appreciated as we don't have here Mac machines to test it.

nsubiron, 2018-02-06:
Check out https://github.com/carla-simulator/carla/issues/176#issuecomment-363372488

thegleb, 2018-03-31:
I feel like I'm pretty close to having Carla run on my OSX but I can't seem to figure out how to get past this error. Any advice?
```
Undefined symbols for architecture x86_64:
  "FRHICommandListBase::WaitForRHIThreadTasks()", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
  "_GRHICommandList", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
      ASceneCaptureCamera::WritePixels(float) const in Module.Carla.cpp.o
  "FRHICommandListBase::WaitForDispatch()", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
  "FRHICommandListExecutor::ExecuteList(FRHICommandListImmediate&)", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
  "FlushRenderingCommands()", referenced from:
      FServerGameController::Tick(float) in Module.Carla.cpp.o
  "_GIsRunningRHIInSeparateThread_InternalUseOnly", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
  "_GMainThreadBlockedOnRenderThread", referenced from:
      ASceneCaptureCamera::Tick(float) in Module.Carla.cpp.o
  "_GIsThreadedRendering", referenced from:
      ASceneCaptureCamera::Tick(float) in Module.Carla.cpp.o
      FRenderCommand::GetDesiredThread() in Module.Carla.cpp.o
  "_GMaxRHIShaderPlatform", referenced from:
      ASceneCaptureCamera::Tick(float) in Module.Carla.cpp.o
  "FRHICommandListBase::WaitForTasks(bool)", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
  "_GDynamicRHI", referenced from:
      ASceneCaptureCamera::WritePixelsNonBlocking(float, FRHICommandListImmediate&) const in Module.Carla.cpp.o
      ASceneCaptureCamera::WritePixels(float) const in Module.Carla.cpp.o
  "GetImmediateCommandList_ForRenderCommand()", referenced from:
      ASceneCaptureCamera::Tick(float) in Module.Carla.cpp.o
      TGraphTask<ASceneCaptureCamera::Tick(float)::EURCMacro_FWritePixelsNonBlocking>::ExecuteTask(TArray<FBaseGraphTask*, FDefaultAllocator>&, ENamedThreads::Type) in Module.Carla.cpp.o
      TGraphTask<ASceneCaptureCamera::Tick(float)::EURCMacro_FWritePixels>::ExecuteTask(TArray<FBaseGraphTask*, FDefaultAllocator>&, ENamedThreads::Type) in Module.Carla.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ERROR: UBT ERROR: Failed to produce item: /Users/gleb/dev/deep-learning-learning/carla/Unreal/CarlaUE4 4.18/Plugins/Carla/Binaries/Mac/UE4Editor-Carla.dylib
Total build time: 26.00 seconds (Local executor: 0.00 seconds)
Command /Users/gleb/UnrealEngine_4.18/Engine/Build/BatchFiles/Mac/Build.sh failed with exit code 5
```
[image]

thegleb, 2018-04-05:
Update: got it to complete the build by modifying the `Carla.build.cs` further:
```
    PublicDependencyModuleNames.AddRange(
      new string[]
      {
        "Core",
        // ... add other public dependencies that you statically link with here ...
        "RHI", // added this
        "RenderCore" // and this
      }
      );
```

analog-cbarber, 2018-07-04:
I got this working using Xcode 8.2.1, which is based on clang 3.9. I had to modify the scripts to use `clang`/` clang++` without the version suffix. I did run into one problem that I do not understand:
the Makefile generated by the GenerateProjectFiles script is targeted at Linux instead of Mac :

~~~Makefile
BUILD = bash "$(UNREALROOTPATH)/Engine/Build/BatchFiles/Linux/Build.sh"
PROJECTBUILD = mono "$(UNREALROOTPATH)/Engine/Binaries/DotNET/UnrealBuildTool.exe"
...
UE4Editor:
         $(BUILD) UE4Editor Linux Development  $(ARGS)
~~~

When I manually changed `Linux` to `Mac` and add mono to my PATH, I was able to complete the
build. I cannot figure out how to fix the Makefile generation. Invoking the GenerateProjectFiles.sh
script with `-Platforms=Mac` doesn't do any good. Not sure what I did wrong.

analog-cbarber, 2018-07-10:
It appears that the `-makefiles` flag to the UnrealBuildTool does not properly support Mac targets. The `-cmakefiles` appears to do the right thing, but I have not actually tried doing a build yet. Since it is not clear what version of llvm/clang is used by later releases of Xcode, and the 0.9 release of CARLA is using clang 5.0, I suspect that using cmake will be preferable to Xcode.

nsubiron, 2018-07-10:
@analog-cbarber Thanks for the report!

In Linux we use a very specific version of the compiler because we need to link against libc++ (instead of the usual libstdc++ for Linux). In Mac, I imagine we can go with the default compiler installed like we do on Windows.

And for the sh scripts, they have been written for Linux and only tested on Linux, we may need to adapt them.

analog-cbarber, 2018-07-10:
That's good, so the Xcode build approach may still be ok? So the main risk would be if CARLA started to depend on C++ features in clang 5.0 that are not yet supported in the Xcode variant?

nsubiron, 2018-07-10:
That's likely to happen, in the latest changes in networking code we use C++17 features, most of them already available in previous versions of clang with -std=c++1z but some may not.

analog-cbarber, 2018-07-10:
OK. So either I need to figure out how to build properly using actual llvm/clang 5.0 or hope that Xcode supports the required features. It seems that Xcode 10 may be based on 5.0 (see Joky's comment on this thread: https://news.ycombinator.com/item?id=16545037)

analog-cbarber, 2018-07-11:
According to https://clang.llvm.org/cxx_status.html, it appears that clang 4.0 should support most of the C++17 features. The only ones I see that are missing are *constexpr lambda expressions* ([P0170R1](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0170r1.pdf)) 
and *template argument deduction for class templates* ([P0091R3](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0091r3.html) and [P0512R0](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0512r0.pdf)). If you aren't using either of those features, then I would expect clang++ 4.0 and Xcode 9.2 to work. That would be nice if true, since the official 4.19 release appears to be built with Xcode 9.2.

nsubiron, 2018-07-12:
I don't think we use any of those features, up until recently all our code base was compiling with clang-3.9 and -std=c++14, it's going to be starting at CARLA 0.9.0 that we move to clang-5.0 and C++17. The C++17 standard has some nice features that may help performance, and hopefully soon will be fully supported by most compilers.

So for now shouldn't be any problems compiling. However, we won't be maintaining compatibility as we simply don't have enough people to maintain it.

analog-cbarber, 2018-07-13:
You could still probably get away with using C++17 features that are compatible with whatever runtime UE 4.19 is using. I am not sure which ones that would be...

FWIW, the changes I made to get the Mac build to work using Xcode 8.2.1 are here:

https://github.com/analog-garage/carla/commits/mac-build

I won't issue a pull request, because this is hackier than I would like and will require significant reworking for the 0.9 release in any case.

edufford, 2018-07-15:
Also, for installing the build dependencies, I used:

- LLVM/Clang 3.9
`brew tap llvm-hs/homebrew-llvm`
`brew install llvm-3.9`

- tac command by symbolic link
`brew install coreutils`
`ln -s /usr/local/bin/gtac /usr/local/bin/tac`

- Others
`brew install cmake autoconf automake ninja libtool wget`

- Python 3.6 (installed by Conda)
`pip install requests`

When building the plugins in Unreal Engine, I also had to downgrade errors for unused lambda capture to warnings following https://answers.unrealengine.com/questions/779733/xcode-build-errors-lambda-capturenot-used.html:
1. Modify *Engine/Source/Programs/UnrealBuildTool/Platform/MacToolChain.cs* to add:
    ```
    Result += " -Wno-unused-lambda-capture";
    ```
    after the existing line:
    ```
    Result += " -Wno-unused-value";
    ```
2. Run `xbuild UnrealBuildTool.csproj` (deprecated) or `msbuild UnrealBuildTool.csproj` from the *Engine/Source/Programs/UnrealBuildTool/* directory

raskolnikov-reborn, 2018-08-08:
I get the "UE4Editor-Carla.dylib missing" along with one more. Rebuilding fails and it says try from source manually. Any help with this?

edufford, 2018-08-08:
The missing dylib messages is normal, just needs the UE rebuild to work.  Are you using the latest Carla 0.9.0?  It was updated for Unreal Engine 4.19 instead of 4.18 so there may be new UE rebuild issues.  I haven’t had a chance to try it with UE 4.19 yet.

analog-cbarber, 2018-08-17:
I got a Mac build working based on the current master branch (i.e. 0.9.0) using
Xcode 9.2 and the standard UE 4.19 download, so there is no need to build libc++
or the Unreal Engine.

Some changes were required.

I have pushed what I have so far on the [mac-build-0.9 branch](https://github.com/analog-garage/carla/tree/mac-build-0.9) of our [fork](https://github.com/analog-garage/carla).

See https://github.com/analog-garage/carla/blob/mac-build-0.9/Docs/how_to_build_on_mac.md

edufford, 2018-08-20:
@analog-cbarber Nice work!

I was able to use your fork's **mac-build-0.9** branch to build Carla 0.9 with Xcode 9.4.1 as well, after modifying the Xcode version check in the *Util/BuildTools/Setup.command* file.

Copying the *GenerateProjectFiles.sh* to the UE root folder didn't work for me though, so instead I was able to modify the *Util/BuildTools/BuildCarlaUE4.command* file to call the built in Mac *GenerateProjectFiles.sh* that comes with the UE engine install package in place:

*Util/BuildTools/BuildCarlaUE4.command*
```
log "Generate Unreal project files."

#if [[ ! -f ${UE4_ROOT}/GenerateProjectFiles.sh ]]; then
#  fatal_error "No GenerateProjectFiles.sh in ${UE4_ROOT}. You can copy this from UnrealEngine source tree."
#fi

#${UE4_ROOT}/GenerateProjectFiles.sh -project="${PWD}/CarlaUE4.uproject" -game -engine -makefiles

# HACK! This generates wrong targets on the Mac! Just replace Linux with Mac everywhere:
#sed -i .original -e "s/Linux/Mac/g" Makefile

# GenerateProjectFiles.sh requires to change directory to location of UE engine Mac batch files
pushd "${UE4_ROOT}/Engine/Build/BatchFiles/Mac/" >/dev/null
./GenerateProjectFiles.sh -project="${CARLAUE4_ROOT_FOLDER}/CarlaUE4.uproject" -game -engine -makefiles
popd >/dev/null
```

The Xcode project workspace *Unreal/CarlaUE4/CarlaUE4.xcworkspace* that this generates worked fine to Build & Run the project and launch the UE Editor.  It would be great if your **how_to_build_on_mac.md** doc also included some info about using this file to build/run the project too.

For the command line UE project build though, instead of using make with the Linux->Mac swapped UE Makefile, I had to modify the *Util/BuildTools/BuildCarlaUE4.command* file to use UE's Mac *Build.sh* script directly and add **-NoUBTMakefiles** to prevent an error about invalid makefile contents from the UE Build Tool:

*Util/BuildTools/BuildCarlaUE4.command*
```
log "Build CarlaUE4 project."
#make CarlaUE4Editor

# Build.sh requires to change directory to location of UE engine root
pushd "${UE4_ROOT}" >/dev/null
./Engine/Build/BatchFiles/Mac/Build.sh UE4Editor Mac Development -NoUBTMakefiles
popd >/dev/null
```

For Mac, Xcode is probably the way to go anyways as long as the AppleClang version is still compatible.  Thanks again for posting your work!

analog-cbarber, 2018-08-20:
Did you copy GenerateProjectFiles.sh from the root of the Unreal Engine source tree? I think that should work. I would have checked in a copy of the file I used if not for the conflicting copyright.

analog-cbarber, 2018-08-20:
Yes, it should be possible to build everything using Xcode and plan to add am option for that if not make it the default on the Mac.

edufford, 2018-08-20:
Ah, I was looking for the *GenerateProjectFiles.sh* file in the /UE_4.19/Engine/Source/ folder since I installed it from the Epic Games Launcher instead of from the full source package, so I could only find the file in the /UE_4.19/Engine/Build/BatchFiles/Mac/ folder and copying that one over didn't work.

It did work when I downloaded the *GenerateProjectFiles.sh* file from Epic's source repo as you linked, but it looks like this script is basically just running:

```
cd Engine/Build/BatchFiles/Mac
sh ./GenerateLLDBInit.sh
sh ./GenerateProjectFiles.sh $@
```

So it seems that if the *BuildCarlaUE4.command* file just manually ran these scripts directly, similar to my modification, then there would be no need for copying this file over from the main engine source.

danxuhk, 2018-08-23:
when run python example.py, it shows 'AttributeError: 'module' object has no attribute 'Client' 'in
Traceback (most recent call last):
  File "example.py", line 114, in <module>
    main(add_a_camera=False, enable_autopilot=True)
  File "example.py", line 43, in main
    client = carla.Client('localhost', 2000)
Any idea on this issue?

danxuhk, 2018-08-23:
I am using python 2.7. After the compilation, I copied the folder 'carla' under PythonClient to PythonAPI, and then run python example.py. The issue comes out. Actually which .py file under carla defines the Client class? I am not able to find a definition from there.

analog-cbarber, 2018-08-23:
Not sure where you got the idea to copy that directory.  The new API is mostly C++ now. You need to install the egg in PythonAPI/dist into your python environment using easy_install.

analog-cbarber, 2018-08-23:
I pushed some changes to my fork that allow later versions of Xcode and use Xcode by default to
build everything except boost.

danxuhk, 2018-08-23:
The example.py can be run successfully, while I run python client_example.py, another issue of failed to read data comes out:
INFO: listening to server localhost:2000
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out

Have you ever encountered something like this?

analog-cbarber, 2018-08-23:
I haven't seen that. Are you sure you don't have another program using port 2000?

danxuhk, 2018-08-24:
I am using the port 2000 by default, and I am able to run example.py in PythonAPI. But I couldn't run the sample in PythonClient, showing failed to read data. If example.py could connect to the port 2000, should client_example.py also be able to connect to the same port, right?

marcgpuig, 2018-08-24:
Hey @danxuhk. 
Please do not duplicate here your questions from your issue #697, that has already been answered.
Thanks!

mariaka, 2018-08-26:
Hi @analog-cbarber ! I tried ``` make launch ``` on your fork of carla with XCode Version 9.4.1 , but I get ``` clang 5.0 is required, but it's not installed.
make sure you build Unreal Engine with clang 5.0 too.
make: *** [setup] Error 1 ```. What's the recommended setup here? Should I install llvm via homebrew?

analog-cbarber, 2018-08-26:
You need to be using the mac-build-0.9 branch and you either need to use make or if invoking the build scripts directly you need to use the ones ending in .command

Abanoub-G, 2018-09-24:
Hey,
      I am trying t build CARLA on my Mac as well. I am not a professional programmer. I got a bit lost in what you guys did. Can you just give me a brief list of bullet points of the steps I am meant take.

Many thanks
AG

analog-cbarber, 2018-09-25:
If you want to use my fork, then you should check out either the mac-build-0.8 branch or the mac-build-0.9 branch and read doc/how_to_build_on_mac.md.

The 0.9 build process is a lot easier but I haven't updated the branch.

I suspect that it should be possible to build 0.8 against the regular Unreal 4.18 download, and thus avoid having to build it, but I haven't tried it yet.

Abanoub-G, 2018-09-27:
> If you want to use my fork, then you should check out either the mac-build-0.8 branch or the mac-build-0.9 branch and read doc/how_to_build_on_mac.md.
>
> The 0.9 build process is a lot easier but I [...]

Hi,
    Thanks a lot for your reply. I have been following your doc/how_to_build_on_mac.md but then I got stuck at running the ./Update.sh file. I get this error:

Backing up existing Content...
/Users/abanoubghobrial/carla/Unreal/CarlaUE4/Content/Carla -> /Users/abanoubghobrial/carla/Unreal/CarlaUE4/Content/Carla_20180928004455
Traceback (most recent call last):
  File "./Util/download_from_gdrive.py", line 15, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

When I try to install requests by running " pip install requests ", this what I get in return:

Requirement already satisfied: requests in /usr/local/lib/python2.7/site-packages (2.5.1)

Which means it is already installed. Any idea what I am doing wrong here?

Abanoub-G, 2018-09-28:
> > If you want to use my fork, then you should check out either the mac-build-0.8 branch or the mac-build-0.9 branch and read doc/how_to_build_on_mac.md.
> > The 0.9 build process is a lot easier but I haven't updated the branch.
> > I suspect that it should be possible to build 0.8 against the regular Unreal 4.18 download, and thus avoid having to build it, but I haven't tried it yet.
> 
> Hi,
> Thanks a lot for your reply. I have been following your doc/how_to_build_on_mac.md but then I got stuck at running the ./Update.sh file. I get this error:
> 
> Backing up existing Content...
> /Users/abanoubghobrial/carla/Unreal/CarlaUE4/Content/Carla -> /Users/abanoubghobrial/carla/Unreal/CarlaUE4/Content/Carla_20180928004455
> Traceback (most recent call last):
> File "./Util/download_from_gdrive.py", line 15, in 
> import requests
> ModuleNotFoundError: No module named 'requests'
> 
> When I try to install requests by running " pip install requests ", this what I get in return:
> 
> Requirement already satisfied: requests in /usr/local/lib/python2.7/site-packages (2.5.1)
> 
> Which means it is already installed. Any idea what I am doing wrong here?

Alright, requests actually exits on my computer but I had to run it using python.
So what I did is:
run python on terminal by typing " python " 
then " import requests"
then exit the python environment  and run ./Update.sh

analog-cbarber, 2018-09-28:
That seems strange. Glad you got it working.

Abanoub-G, 2018-09-28:
> I got a Mac build working based on the current master branch (i.e. 0.9.0) using
> Xcode 9.2 and the standard UE 4.19 download, so there is no need to build libc++
> or the Unreal Engine.
>
> Some changes [...]

Hi, 
What is the difference between the "mac-build-0.9 branch" and the "fork"?  They both seem to be having the same files...

Abanoub-G, 2018-09-28:
Hi, 
    also in your guide, in the bit where it says "_In order for boost-python to build properly, you may need to add a user-config.jam file to your home directory describing the location of your python implementations. For instance, to use the preinstalled python2 and a python 3.6 from python.org installer you might use:_"...   I am not really sure how to make this .jam file and where do I place it?

analog-cbarber, 2018-09-28:
Uh, the link I gave is for the mac-build-0.9 branch, so of course it is the same as itself! The "fork" refers to my fork of the CARLA git repository, which contains all of their branches, plus my mac build branches. You will notice that the mac-build- branches differ from the respective master branches from which they were derived. Note I have not updated my fork since I did this work.

Regarding the `user-config.jam` file, you put it in your home directory, just like I said i, and I gave an example of what it looks like. Just change the paths to match your configuration. If that doesn't work for you, you should look at the Boost documentation (e.g. https://www.boost.org/doc/libs/1_67_0/libs/python/doc/html/building/configuring_boost_build.html)

Abanoub-G, 2018-09-28:
> Uh, the link I gave is for the mac-build-0.9 branch, so of course it is the same as itself! The "fork" refers to my fork of the CARLA git repository, which contains all of their branches, plus my [...]

Hi,
   I have used "nano user-config.jam" to create the file the content you mentioned. However doing this didn't make any difference I am still getting the error below when I run "./Update.sh":

Backing up existing Content...
/Users/abanoubghobrial/carla-master/Unreal/CarlaUE4/Content/Carla -> /Users/abanoubghobrial/carla-master/Unreal/CarlaUE4/Content/Carla_20180928175042
Traceback (most recent call last):
  File "./Util/download_from_gdrive.py", line 15, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

I have tried importing requests but that's what I get in return:

-bash: import: command not found

simply because "import" is a python command.

I genuinely don't know what to do now. A step by step guide with screenshots would be really appreciated or a video guide. I think it would benefit all of the future users. Especially people who are not programming experts like myself.

analog-cbarber, 2018-09-28:
Sorry, I don't have time for that, and there is obviously no guarantee that everyone's machine is configured the same way. I have no idea if it matters what version of Mac OSX you have or what specific python installs you have and don't have time to figure that out. I don't work on the CARLA project. I just posted what I did for my own work. I cannot really help you with problems that I did not encounter myself when I did this.

Having said that, it looks like that is failing to download the CARLA game content, which is not part of this repository. Perhaps something changed with that step that is no longer compatible with the original 0.9 version. I really don't know.

I should also say that doing anything non-trivial with CARLA will require programming, so you will have to get comfortable with that. If you want the most pain free experience, I would recommend that you switch to a supported platform and stick to the stable 0.8 version of CARLA.

Abanoub-G, 2018-10-03:
Right, when I do make launch. It gives me a fatal error that it can find the path boost/... etc. 
I know where the file boost is in the Carla folder but the path for the folder in the .h files is put between "< >". Which makes the C++ code look for the files in the system directories of my Mac. 

So my problem is which files on my Mac are classified as the "system directories". i.e. when I out a path between < > in C++ which directories on my Mac would it go looking in?

Thanks

analog-cbarber, 2018-10-16:
FYI, I updated the mac-build-0.8 branch on my fork (https://github.com/analog-garage/carla/tree/mac-build-0.8) to use Xcode 9.2 and and out-of-the-box UE 4.18, so you no longer have to build the Unreal engine or llvm.

dano-ai, 2019-02-01:
Hi @analog-cbarber 
Thanks for your custom branches. I managed to maneuver a few hurdles through 'make launch' and bumped into a brick wall now.

If I run the following code:
`./Engine/Build/BatchFiles/Mac/Build.sh ShaderCompileWorker Mac Development -verbose
`

```
VERBOSE: Skipping project file generator registration for Win32 due to no valid BuildPlatform.
VERBOSE: Skipping project file generator registration for Win64 due to no valid BuildPlatform.
Creating makefile for ShaderCompileWorker (no existing makefile)
==============================================================================
ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /Users/Shared/EpicGames/UE_4.21/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:

       (see ../Programs/UnrealBuildTool/Log.txt for full exception trace)

BuildException: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
Location: /Users/Shared/EpicGames/UE_4.21/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
Target rules found:

  at UnrealBuildTool.RulesAssembly.CreateTargetRules (System.String TargetName, UnrealTargetPlatform Platform, UnrealTargetConfiguration Configuration, System.String Architecture, Tools.DotNETCommon.FileReference ProjectFile, UnrealBuildTool.ReadOnlyBuildVersion Version, System.String[] Arguments, Tools.DotNETCommon.FileReference& TargetFileName) [0x00000] in <filename unknown>:0
  at UnrealBuildTool.RulesAssembly.CreateTargetRules (System.String TargetName, UnrealTargetPlatform Platform, UnrealTargetConfiguration Configuration, System.String Architecture, Tools.DotNETCommon.FileReference ProjectFile, UnrealBuildTool.ReadOnlyBuildVersion Version, System.String[] Arguments, Tools.DotNETCommon.FileReference& TargetFileName) [0x00000] in <filename unknown>:0
  at UnrealBuildTool.UEBuildTarget.CreateTarget (UnrealBuildTool.TargetDescriptor Desc, System.String[] Arguments, Boolean bSkipRulesCompile, Boolean bCompilingSingleFile, Boolean bUsePrecompiled, UnrealBuildTool.ReadOnlyBuildVersion Version) [0x00000] in <filename unknown>:0
  at UnrealBuildTool.UnrealBuildTool.RunUBT (UnrealBuildTool.BuildConfiguration BuildConfiguration, System.String[] Arguments, Tools.DotNETCommon.FileReference ProjectFile, Boolean bCatchExceptions) [0x00000] in <filename unknown>:0
==============================================================================

```
Mojave 10.14.2, Unreal 4.21.2 and XCode 10.1

FYI, in the source directory, there are 2 cs files:
UE4Editor.Target.cs
UE4Game.Target.cs

Can you please shed some light on this issue? I will greatly appreciate it.

dano-ai, 2019-02-01:
I also found out that there are more errors before the one above:

BuildCarlaUE4.command: Generate Unreal project files.

```
Setting up Unreal Engine 4 project files...

GenerateProjectFiles ERROR: It looks like youre missing some files that are required in order to generate projects.  Please check that youve downloaded and unpacked the engine source code, binaries, content and third-party dependencies before running this script.
BuildCarlaUE4.command: Build CarlaUE4 project.
2019-02-02 01:58:08.129 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Config/BaseEditorLayout.ini" failed, errno = 2
2019-02-02 01:58:09.350 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Plugins/Experimental/AlembicImporter/Source/ThirdParty/Alembic/alembic/houdini/SOP_AlembicIn/OPalembic/Object_1alembicarchive/Help" failed, errno = 2
2019-02-02 01:58:09.417 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Plugins/Experimental/AlembicImporter/Source/ThirdParty/Alembic/alembic/houdini/SOP_AlembicIn/OPalembic/Object_1alembicxform/Help" failed, errno = 2
2019-02-02 01:58:24.527 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Plugins/Experimental/AlembicImporter/Source/ThirdParty/Alembic/hdf5/tools/testfiles/tmulti-g.h5" failed, errno = 2
2019-02-02 01:58:24.824 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Plugins/Experimental/AlembicImporter/Source/ThirdParty/Alembic/hdf5/tools/testfiles/tnoddlfile.ddl" failed, errno = 2
2019-02-02 01:58:26.344 xcodebuild[59304:15386536] +dataWithFirstBytes:1024 ofFile:"/Users/Shared/EpicGames/UE_4.21/Engine/Plugins/Experimental/AlembicImporter/Source/ThirdParty/Alembic/hdf5/tools/testfiles/twithddlfile.ddl" failed, errno = 2
2019-02-02 01:58:37.284 xcodebuild[59304:15386541]  DTDeviceKit: deviceType from 819db3401a69f4c350e8813f2db3a81926d0924a was NULL
Prepare build
note: Using legacy build system
```

This might be related.

dano-ai, 2019-02-01:
@edufford can you please take a look at this error? Thanks in advance.

```
sh ./GenerateProjectFiles.sh $@

Setting up Unreal Engine 4 project files...

GenerateProjectFiles ERROR: It looks like youre missing some files that are required in order to generate projects.  Please check that youve downloaded and unpacked the engine source code, binaries, content and third-party dependencies before running this script.
```

edufford, 2019-02-05:
@dano-ai Sorry, I don't have the compute environment set up for this anymore, but if I have a chance to check into it again later I'll let you know.  Best of luck!

analog-cbarber, 2019-02-17:
I have been working a bit on this the last week or two and have managed to get a fairly recent version of the master branch to build on my Mac running High Sierra (10.13.6) using a downloaded Unreal 4.21.2 and using Xcode 9.4 (which is the version of Xcode that Epic claims is used to build UE). However, I still have significant problems:

* The CarlaUE4 game freezes up my machine (an 2015 era Macbook Pro) when I run the packaged game or try to run in standalone mode. I can run it in PIE mode. My efforts at tweaking the Mac configuration settings have only made things worse.

* When I import carla in python3 I get the error:

    ~~~
    >>> import carla
   Traceback (most recent call last):
     File "<stdin>", line 1, in <module>
     File "/Users/cbarber/miniconda3/envs/carla/lib/python3.6/site-packages/carla-0.9.3-py3.6- 
  macosx-10.7-x86_64.egg/carla/__init__.py", line 7, in <module>
       from .libcarla import *
   ImportError: dlopen(/Users/cbarber/miniconda3/envs/carla/lib/python3.6/site-packages/carla-0.9.3-py3.6-macosx-10.7-x86_64.egg/carla/libcarla.cpython-36m-darwin.so, 2): Symbol not found: _TIFFClose
     Referenced from: /Users/cbarber/miniconda3/envs/carla/lib/python3.6/site-packages/carla-0.9.3-py3.6-macosx-10.7-x86_64.egg/carla/libcarla.cpython-36m-darwin.so
        Expected in: flat namespace
    in /Users/cbarber/miniconda3/envs/carla/lib/python3.6/site-packages/carla-0.9.3-py3.6-macosx-10.7-x86_64.egg/carla/libcarla.cpython-36m-darwin.so
    ~~~

    so I still have some build dependencies to work out.

I would love to get the game to work in standalone mode, but would at least like to get the python client working.

I have checkpointed my recent changes on the mac-build-0.9 branch of my fork.

analog-cbarber, 2019-02-17:
I noticed that I was missing library commands for mac in my setup.py. When I added them and rebuilt, I was able to use the `spawn_npc.py` script to add agents to CARLA running in PIE mode.

I pushed the change to my mac-build-0.9 branch.

analog-cbarber, 2019-02-17:
`manual_control.py` fails because pygame doesn't seem to be able to find any fonts on my mac. I worked around this by manually adding a list of fonts to look for and am now able to run the manual control program.

Not sure what to do about the freezing problem in standalone mode. When this happens I have to powercycle my mac. Not sure how to debug this. Perhaps there is a log file somewhere that will give me a clue...

analog-cbarber, 2019-03-02:
I merged 0.9.4 into [my fork](https://github.com/analog-garage/carla/tree/mac-build-0.9) and built.

Haven't tested other than to drive the manual control a bit connected to the game running in PIE mode. I have no reason to believe that the freeze problem has gone away. (Speaking of which, I have tried running with a variety of different settings but have not found any that work. There are some GPU logs, but aren't especially informative to me.)

I should say that while it is very useful to be able to do development on my carla related code on my Macbook pro, it's GPU isn't powerful enough to make good use of CARLA. Unless someone figures out how to make use of external GPUs, I don't think that Mac will ever be a recommended platform for working with CARLA. If the network overhead is not too bad, it might be more generally useful to be able to remotely drive a CARLA simulation from a Mac, in which case it would make sense to support the client API.

WolfgangFahl, 2019-07-21:
I get "OpenGL" Not Supported when trying to run Unreal Engine 4.22.3
[image]
https://answers.unrealengine.com/questions/530680/what-does-opengl-not-supported-mean.html explains this. So you might want to watch out that your hardware is fit for the current version of the Unreal Engine

WolfgangFahl, 2019-07-22:
At http://wiki.bitplan.com/index.php/Car_simulation_software#Carla I have documented my attempts in compiling for a Mac with mixed results. On my 10.13.6 laptop I finally got it working after many hours of fiddling.

WolfgangFahl, 2019-07-22:
make package fails with:

bash: /Users/Shared/EpicGames/UE_4.21/Engine/Build/BatchFiles/Linux/RunMono.sh: No such file or directory

while there is RunMono.sh is in Mac directory. Why is this not selected?

analog-cbarber, 2019-07-22:
You shouldn't even bother with make package until you can launch the game in the editor and run it in standalone mode.

Since I was unable to do that (because my Mac froze), I probably never tried 'make package' and would not have noticed if it was not correct.

rsatyan, 2019-09-13:
In file included from /Users/savatara/Downloads/carla/LibCarla/source/carla/opendrive/OpenDrive.h:9:
/Users/savatara/Downloads/carla/LibCarla/cmake/../source/carla/Memory.h:9:10: fatal error:
'boost/enable_shared_from_this.hpp' file not found
#include <boost/enable_shared_from_this.hpp>
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Keep getting this error, I checked the file exists in the boost dir. I'm stuck. Please help!

rsatyan, 2019-09-19:
Having an issue building PythonAPI.3 - Have followed the instructions in how to build for mac. 

In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:7:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/Episode.h:9:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/AtomicSharedPtr.h:9:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4448:28: error:
      cannot cast 'element_type' (aka 'carla::client::detail::EpisodeState') to its private base class 'const
      enable_shared_from_this<carla::client::detail::EpisodeState>'
    __r.__enable_weak_this(__r.__ptr_, __r.__ptr_);
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4810:29: note: in
      instantiation of function template specialization
      'std::__1::shared_ptr<carla::client::detail::EpisodeState>::make_shared<const unsigned long long &>' requested here
    return shared_ptr<_Tp>::make_shared(_VSTD::forward<_Args>(__args)...);
                            ^
/Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:29:19: note: in instantiation of
      function template specialization 'std::__1::make_shared<carla::client::detail::EpisodeState, const unsigned long long &>'
      requested here
      _state(std::make_shared<EpisodeState>(info.id)),
                  ^
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:7:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/Episode.h:15:
/Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/EpisodeState.h:25:7: note:
      implicitly declared private here
    : std::enable_shared_from_this<EpisodeState>,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:7:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/Episode.h:9:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/AtomicSharedPtr.h:9:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4448:28: error:
      cannot cast 'element_type' (aka 'const carla::client::detail::EpisodeState') to its private base class 'const
      enable_shared_from_this<carla::client::detail::EpisodeState>'
    __r.__enable_weak_this(__r.__ptr_, __r.__ptr_);
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4810:29: note: in
      instantiation of function template specialization 'std::__1::shared_ptr<const
      carla::client::detail::EpisodeState>::make_shared<const carla::sensor::data::RawEpisodeState &>' requested here
    return shared_ptr<_Tp>::make_shared(_VSTD::forward<_Args>(__args)...);
                            ^
/Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:47:26: note: in instantiation of
      function template specialization 'std::__1::make_shared<const carla::client::detail::EpisodeState, const
      carla::sensor::data::RawEpisodeState &>' requested here
        auto next = std::make_shared<const EpisodeState>(CastData(*data));
                         ^
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/source/carla/client/detail/Episode.cpp:7:
In file included from /Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/Episode.h:15:
/Users/savatara/Downloads/carla-mac-build-0.9/LibCarla/cmake/../source/carla/client/detail/EpisodeState.h:25:7: note:
      implicitly declared private here
    : std::enable_shared_from_this<EpisodeState>,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.

baili-zns, 2019-10-15:
> I merged 0.9.4 into [my fork](https://github.com/analog-garage/carla/tree/mac-build-0.9) and built.
>
> Haven't tested other than to drive the manual control a bit connected to the game running in PIE [...]

the 0.9.4 client don't work well and it's greatful if you can help build the 0.9.5 branch

piprrr, 2019-10-15:
> Also, for installing the Mac build dependencies, I used:
[...]
> * tac command by symbolic link
>   `brew install coreutils`
>   `ln -s /usr/local/bin/gtac /usr/local/bin/tac`
[...]

As a small aside, `tac` came over naturally when I did `brew install coreutils`, so I didn't need to do the symlink.  Thank you for the helpful info, though!

raeed17, 2020-01-23:
If someone got CARLA to work on Mac using Xcode could they maybe put in on GitHub for people to use? I feel like it would help a lot of people! Unless this is already available somewhere?

[20 more replies]