carla-simulator/carla #151: segmentation label shows all '0'

sufengniu, 2018-01-20:
Hello, I got image segmentation label issue in Carla, but haven't found any one mentioned it. Not sure if it is a bug or not.

I am using Ubuntu 16, I directly download the version 0.7.0 binaries from website. I add the image segmentation in the sensor data as:
`camera2 = Camera('CameraSeg', PostProcessing='SemasdsnticSegmentation')` 
and store the segmentation ground truth into my local directory. But the segmentation image is all black, and I saw the server terminal confirmed that the segmentation is enabled.

Anyone encountered the similar issues? Thank you!

nsubiron, 2018-01-21:
Hi @sufengniu,

The server encodes the segmentation label in the red channel as a number from 0 to 12, so it looks black at first sight.

In Python you can convert this labels using `carla.image_converter` to get the CityScapes palette. But if you have to convert a bulk of images the best way would be to compile and use the app at "Util/ImageConverter" as it is much faster converting lots of images. Just store all the images in a folder as you do now and run the image converter

```
$ cd Util/ImageConverter
$ make
$ ./bin/image_converter -c semseg -i path/to/image/folder -o output/folder
```

sufengniu, 2018-01-21:
Hi @nsubiron 

Thank you very much, I tried the converter as you mentioned, however, the extracted segmentation image looks like the following, which is very noisy:
[image]

Is this the normal results?

Thank you

nsubiron, 2018-01-21:
Oh well, that's interesting... The resulting image should look like the one it prints `manual_control.py` in the pygame window, each class having its own color.

Can you share how are you generating this image?

sufengniu, 2018-01-23:
Thank you, it turns out that the problem is caused from my site, I have a type in SemanticSegmentation which cause the wrong segmentation image. 

After fixing it, I still found that the segementation image is all black. It turns out the problem come from AMD GPU card, but it becomes fully functional after I switched to Nvidia card. I update my AMD card driver to the latest one (4,17.2), however the unreal engine in carla only support 4.17. So the solution is just switch to nvidia card. Thank you, will close the issue.

closed 2018-01-23 by sufengniu

SimonsLiu, 2019-04-03:
> Hi @sufengniu,
>
> The server encodes the segmentation label in the red channel as a number from 0 to 12, so it looks black at first sight.
>
> In Python you can convert this labels using [...]

I also want to do this conversion for my picture, but I tried that CityScapes palette but picture is still black
Can you give us a brief description of the specific methods or steps? i really want it.