carla-simulator/carla #152: Modifying the Planner

AftermathK, 2018-01-21:
The CARLA platform seems ideal for the current project that our group is working on, but I had two questions in regards to generating a planner using the Modular Pipeline's A* algorithm. 

How do I modify the vehicle's destination and behavior (planner)? Is the traffic random or do they follow predefined paths? It would be useful to control their behavior and destinations as well.

Thank you.

felipecode, 2018-01-22:
THe

xfqbuaa, 2018-01-22:
This is a common question if user want to control the vehicle in simulator.

AftermathK, 2018-01-22:
@xfqbuaa could you please point me in the right direction?

xfqbuaa, 2018-01-29:
This is also a question for me now. 
The Carla idea is different with others simulator, there are a lot of episode and target location to reach. However I did not find any information to get this target location.
In the planner.py ` def get_next_command(self, source, source_ori, target, target_ori):`  target is used in functiob get_next_command.

felipecode, 2018-01-30:
Hello @xfqbuaa and @AftermathK . Sorry for the late answer. 
All the non player cars follow predefined fixed paths defined by a random seed.  However, adding the capability to control multiple vehicles is totally in line with our roadmap :).  The planner is very basic. It is some intermediate thing until we design a better solution. But basically it is as @xfqbuaa says. def get_next_command(self, source, source_ori, target, target_ori)
This function can be used to interact with the planner. You basically send a source position in world coordinates with its orientation  and also a target. It returns you the next high level command ( straight, left, right). Inside this commands are generated from  from a route planned inside a grid.

About @xfqbuaa question. The target positions are controlled by the users. You can define your targets and check if the agent has reach them. Remember that using the targets from the same list as the possible start positions is recommended.

xfqbuaa, 2018-02-02:
Thanks @felipecode for your kindly reply. 

From my side, I spent some time on Carla documents but still have no idea to know Carla design. 
If possible, a simple tutorial to show how to navigate the vehicle in Carla is appreciated. 
More please seen https://github.com/carla-simulator/carla/issues/194
Thanks a lot.

closed 2018-02-06 by felipecode