carla-simulator/carla #153: Pedestrians

AftermathK, 2018-01-21:
I was under the impression that pedestrians were not supported due to licensing issues, has this been resolved or is there a way to purchase my own license?

Thank you.

nsubiron, 2018-01-21:
Hi @AftermathK,

Pedestrians are only available in the compiled version ([latest release here](https://github.com/carla-simulator/carla/releases/latest)) but not when you build from source. The license terms do not allow us to distribute a modifiable copy of the pedestrian models, thus we cannot include them in the open-source project.

AftermathK, 2018-01-21:
Hi @nsubiron,

Thank you for your reply. If I wanted to build from source, is there a way I can integrate them on my own? Were these part of an UE4 package?

TheNihilisticRobot, 2018-01-22:
Hi! @AftermathK 

We are currently working on the issue (Creating some models ourselves, Free of any licensing problem) and we expect to have them as soon as we can in the next two weeks. 
Unfortunately by taking our pedestrians down we've also taken down the system to spawn them and control them (To avoid system errors) so you won't be able to integrate your own pedestrians unless you build also the Pedestrian AI

closed 2018-02-01 by carla-admin