carla-simulator/carla #156: CARLA Performace too low

YashBansod, 2018-01-23:
I am running a Carla Server-Client session on my machine with the following specs:
- Intel Xeon E5-2690 v4
- 64 GB DDR4 RAM
- Nvidia GTX Titan Xp

I have enable three cameras with resolution 640x480 with following post-processing:
- SceneFinal
- Depth
- SemanticSegmentation

For Carla Server I did:
./CarlaUE4.sh /Game/Maps/Town01 -carla-server -windowed -ResX=640 -ResY=480

And the Client is just a modification of client_example.py provided along with the precompiled CARLA.

I am getting just 3 fps on the simulator and the following are my system utilizations:
- Around 50 - 110 % single thread CPU utilization (trough top, this is just single thread utilization since all of my other threads are hardly utilized 1%)
- Around 1500 MB of VRAM and 30 - 40 % GPU Utilization.

As it is clear that there is a lot of compute that my system can provide but CARLA is not utilizing (or maybe this is a problem of UE4), is there any way I can improve my FPS (by making it use all the sytem resouces etc.)?

What is the possible reason for this bottleneck and any suggestions for improving the performance?

curantil, 2018-01-23:
I noticed that saving the images slows the simulation down a lot. I have not fully investigated, but I think that because the disk IO is relatively slow and the writing is done synchronously, the whole simulation is bogged down.

nsubiron, 2018-01-23:
Yes, CARLA server is currently slow, please check issue #10.

nsubiron, 2018-01-23:
@curantil 

> disk IO is relatively slow and the writing is done synchronously

Yes definitively, writing the images to disk is very slow. You can use a RAM drive to speed up or implement some asynchronous client on top of our API. In our examples everything on the client-side is synchronous for simplicity.

YashBansod, 2018-01-24:
Okay, I disabled the writing to disk IO and disabled all image conversion operations.

I ran it again and I got 6-7 FPS (a 100% improvement in performance). This was expected since disk IO was consuming a lot of time. 

But still 6-7 FPS on a machine with such high specifications and such low system utilizations is really unacceptable. 

I profiled the code by running an episode for 506 frames and found that:
<method 'recv' of '_socket.socker' objects>

which I think is:
data = self._socket.recv(length)
in 'tcp.py' itself is consuming >96% of the compute time.

It was called 3106 times for creating 506 frames, and took 77516 ms in total. 
So ~= 25ms per call and ~=153ms per frame.

To my understanding this is just the code to read the sensor data from the server. Am I correct?
If Yes, why is it called 3106 times if only 506 frames were processed.

As Its evident this is the part of code that is slowing the whole thing down. Can someone from the CARLA team please explain what  exactly is happening. Also, if there is anything that I can do to improve the performance.

YashBansod, 2018-01-24:
Finally I changed the settings to run it asynchronously hoping I get a better performance, but it was almost the same. (Slightly better average performance on average ~=23ms per call).

nsubiron, 2018-01-24:
> data = self._socket.recv(length)
> in 'tcp.py' itself is consuming >96% of the compute time.

`socket.recv` is basically waiting for the server to produce the measurements and images, the bottleneck is in the server. Unfortunately that's about the speed you get when you add cameras to the vehicle.

closed 2018-02-01 by carla-admin