carla-simulator/carla #157: 16-bit images

biboffff, 2018-01-23:
Hello,

I want to use carla to render 16-bit images. May someone tell me, what's the easiest way to achieve this?

I tried to set "bUseHDRDisplayOutput=True" in ./Unreal/CarlaUE4/Saved/Config/Linux/GameUserSettings.ini (of the built from source version) and rebuild Carla, but this shows no effect. 

I appreciate any help.

Chris

nsubiron, 2018-01-23:
I believe in the build-from-source version the config files should be under "Unreal/CarlaUE4/Config/", but you can usually test it too in the release version by adding the file to "CarlaUE4/Saved/Config/LinuxNoEditor/".

nsubiron, 2018-01-23:
Hi @biboffff, sorry I overlooked your question. If you mean to retrieve HDR captures at the client side, that's not currently possible. This is currently hard-coded and even if Unreal generates the images in HDR we are going to convert them to an RGBA array before sending to the client.

biboffff, 2018-01-23:
Hello!

Alright, thanks for your response and your help!

Best regards,
Chris

closed 2018-01-25 by marcgpuig

ztysdu, 2022-10-13:
Hi, I'm recently researching how to generate HDR images with carla, and I only found this [URL](https://docs.unrealengine.com/4.26/en-US/RenderingAndGraphics/HDRDisplayOutput/) and your question on the web. Can you tell me how you set up HDR mode? Thank you very much!