carla-simulator/carla #164: Problem running Carla for the first time

eds89, 2018-01-23:
Hello,
When trying to run Carla for the first time, I'm getting the following error.

Does anybody have any advice on this? The OpenGL-compatible graphics driver has been installed. There is plenty of GPU memory available (about 16 GB). It is a 12-core virtual machine.

Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(325017572) SRandInit(325017572).
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.071240
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin Carla
[33mLogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
[0mLogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 20
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 28
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 0
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
[... 180 lines cut ...]
[2018.01.23-10.47.21:026][  0]CrashReportClientLog: State change from SendingFiles to Finished
[2018.01.23-10.47.21:496][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2018.01.23-10.47.21:496][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2018.01.23-10.47.21:529][  0]CrashReportClientLog: Final state (Receiver) = Finished
[2018.01.23-10.47.21:529][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2018.01.23-10.47.21:529][  0]LogExit: Preparing to exit.
[2018.01.23-10.47.21:530][  0]LogExit: Object subsystem successfully closed.
[2018.01.23-10.47.21:532][  0]LogModuleManager: Shutting down and abandoning module HTTP (8)
[2018.01.23-10.47.21:543][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2018.01.23-10.47.21:543][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2018.01.23-10.47.21:543][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2018.01.23-10.47.21:545][  0]LogExit: Exiting.
394cf6a  PlatformInitOpenGL() [Runtime/Core/Public/Containers/ContainerAllocationPolicies.h, line 356]
[Callstack]  11  0x00000000039680ee  FOpenGLDynamicRHI::FOpenGLDynamicRHI() [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/OpenGLDrv/Private/Linux/OpenGLLinux.cpp, line 301]
[Callstack]  12  0x0000000003967bec  FOpenGLDynamicRHIModule::CreateRHI(ERHIFeatureLevel::Type) [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLDevice.cpp, line 1066]
[Callstack]  13  0x00000000039cb88d  PlatformCreateDynamicRHI() [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/RHI/Private/Linux/LinuxDynamicRHI.cpp, line 53]
[Callstack]  14  0x00000000039cb0fd  RHIInit(bool) [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp, line 178]
[Callstack]  15  0x0000000002458b4c  FEngineLoop::PreInit(wchar_t const*) [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp, line 1713]
[Callstack]  16  0x0000000002463b88  GuardedMain(wchar_t const*) [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 42]
[Callstack]  17  0x00000000057f9f19  CommonLinuxMain(int, char**, int (*)(wchar_t const*)) [/home/nsubiron/UnrealEngine_4.17/Engine/Source/Runtime/Linux/LinuxCommonStartup/Private/LinuxCommonStartup.cpp, line 236]
[Callstack]  18  0x00007f4bb3868830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f4bb3868830]
[Callstack]  19  0x0000000002455d99  /home/esantos/Downloads/Carla/CarlaUE4/Binaries/Linux/CarlaUE4(_start+0x29) [0x2455d99]

[2018.01.23-10.47.19:428][  0]LogExit: Executing StaticShutdownAfterError
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

nsubiron, 2018-01-23:
This seems to be the issue

> Unable to dynamically load libGL: GLX is not supported

tlind, 2018-01-24:
Are you trying to run this in some kind of virtual machine or Docker container?
Does e.g. `glxgears` run on your computer, or do you get the same error message?

eds89, 2018-01-24:
Hello,
Yes, I am. I'm trying to get it to run on a Tesla 100 virtual machine.

So, when I run the glxgears command, I get the following:

ubuntu@gpu-carlasim2:~$ glxgears
Xlib:  extension "GLX" missing on display ":0.0".
Error: couldn't get an RGB, Double-buffered visual

And when I run the glxinfo command:

ubuntu@gpu-carlasim2:~$ glxinfo
name of display: :0.0
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Error: couldn't find RGB GLX visual or fbconfig
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".

Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".
Xlib:  extension "GLX" missing on display ":0.0".

Apparently, OpenGL isn't installed, however I can't find a reliable way to
install it. It is supposed to come with the NVIDIA drivers, which are
already installed. Nvidia drivers version 387.34.

felipecode, 2018-01-24:
@eds89 
It seems that you don't have your X setup.
try running:
`sudo nvidia-xconfig -a --use-display-device=None --virtual=1280x1024`
We are currently working in a tutorial for running CARLA on servers ( Untested yet)
https://github.com/carla-simulator/carla/blob/tutorial_for_running_headless/Docs/carla_on_server.md

eds89, 2018-01-24:
> for_running_headless/Docs/carla_on_server.md

Hi,
I actually have a display (lubuntu).

felipecode, 2018-01-24:
Maybe try just the 
sudo nvidia-xconfig ? 

Your problem shows that either drivers or opengl are not properly installed. But it is also can mean that you don't have your x configured to work with the drivers you have. 
I had a similar problem than you. But running on a server without display.

nsubiron, 2018-01-24:
@eds89 also you might try checking the Unreal Engine forums to see if there are common issues running on Lubuntu. Unreal Engine's Linux support is still pretty limited to few distributions.

closed 2018-02-01 by carla-admin

nsubiron, 2018-09-05:
Linking #757.

hsezhiyan, 2019-12-17:
@felipecode I have the same issue. I tried:

- Running the `nvidia-xconfig` command
- Upgrading to a larger GPU
- Updating device drivers to the latest by Nvidia

None of these seem to work. Any suggestion?