carla-simulator/carla #169: TCPConnectionError

coolKeen, 2018-01-25:
Hi,In what situation, the error below may happend?Thanks in advance!

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "client_example_for_shanbo.py", line 164, in send_measurement
    measurements,_ = client.read_data()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/client.py", line 117, in read_data
    data = self._stream_client.read()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 74, in read
    header = self._read_n(4)
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 92, in _read_n
    raise TCPConnectionError(self._logprefix + 'connection closed')
carla.tcp.TCPConnectionError: (localhost:2001) connection closed

nsubiron, 2018-01-25:
Hi @coolKeen,

Pretty much what it says "connection closed" :) something happened and the server got disconnected, only with this I can't tell you more, could be just a network error. Does it happen often to you? Which versions are you using (client and server)?

Btw we encapsulate any exception in the connection as "TCPConnectionError" so it is possible to catch this specific type of exception. You may want to catch this exception in your python client, then reconnect and restart the episode without making the client crash.

coolKeen, 2018-01-25:
emmmmmm.It is a little complex.I want to use carla to test my code which is wrote in cpp.So I use lcm as a message communication tool between the code wrote in cpp and the code wrote in python.Python code send some information to cpp code and cpp code use that information to calculate steer and throttle and send the steer and throttle to python (client) code.But when I use multi-thread.The problem like below occur:
INFO: listening to server localhost:2000
CarlaClient connected
Received weather_id(1) and player_start(34) from channel(ENV_CONFIG)!
Starting new episode...
34
####################
{'location': (56.26, 234.2, 0.22), 'agent_type': 'traffic_light', 'state': '2', 'orientation': (1.0728836059570312e-06, -0.9999999403953552, 0.0)}
{'location': (62.15, 219.92, 3.53), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'location': (77.74, 218.81, 3.55), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'location': (60.88, 244.82, 1.84), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'agent_type': 'player_agent', 'collision_vehicles': 0.0, 'forward_speed': 0.0, 'collision_pedestrians': 0.0, 'acceleration': (0.0, 0.0, 0.0), 'location': (71.0400390625, 237.43, 1.37), 'orientation': (-1.0, -6.4373016357421875e-06, 0.0), 'collision_other': 0.0, 'intersection_offroad': 0.0, 'intersection_otherlane': 0.0}
Received steer(0.0) , throttle(0.0) and on_inflexion(0) from channel(DECISION_INFO)!
send original decision
-5.231584964349167e-06
1.0
####################
set autopilot succeed!
get info around succeed!
{'location': (56.26, 234.2, 0.22), 'agent_type': 'traffic_light', 'state': '2', 'orientation': (1.0728836059570312e-06, -0.9999999403953552, 0.0)}
{'location': (60.88, 244.82, 1.7528887939453126), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'agent_type': 'player_agent', 'collision_vehicles': 0.0, 'forward_speed': -4.38001546143596e-09, 'collision_pedestrians': 0.0, 'acceleration': (6.57002274806473e-08, 4.229321699713434e-13, 0.0), 'location': (71.0400390625, 237.43, 1.3427870178222656), 'orientation': (-1.0, -6.4373016357421875e-06, 0.0), 'collision_other': 0.0, 'intersection_offroad': 0.0, 'intersection_otherlane': 0.0}
####################
Done.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "client_example_for_shanbo.py", line 164, in send_measurement
    measurements,_ = client.read_data()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/client.py", line 117, in read_data
    data = self._stream_client.read()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 74, in read
    header = self._read_n(4)
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 84, in _read_n
    raise TCPConnectionError(self._logprefix + 'not connected')
carla.tcp.TCPConnectionError: (localhost:2001) not connected

^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

My python code like this:
#!/usr/bin/env python3

# Copyright (c) 2017 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB), and the INTEL Visual Computing Lab.
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

"""Basic CARLA client example."""

from __future__ import print_function

import argparse
import logging
import random
import sys
import time
import numpy as np
import lcm
[... 449 lines cut ...]
         cout << "Error:unable to create receive thread," << rc << endl;
         exit(-1);
    }
    rc = pthread_create(&threads[SEND_THREAD],NULL,send_decision,NULL);
    if(rc){
         cout << "Error:unable to create send thread," << rc << endl;
         exit(-1);
    }
    pthread_attr_destroy(&attr);
    rc = pthread_join(threads[RECEIVE_THREAD], &status);
    if (rc){
         cout << "Error:unable to join receive thread " << rc << endl;
         exit(-1);
    }
    rc = pthread_join(threads[SEND_THREAD], &status);
    if (rc){
         cout << "Error:unable to join send thread " << rc << endl;
         exit(-1);
    }
    pthread_mutex_destroy(&around_agent_info_mutex);
    cout<<"Main program exit!"<<endl;
    pthread_exit(NULL);
}

After I run python server,I run python (client)code and then I run the cpp code.The problem above occurs.I use version 0.7.0.

coolKeen, 2018-01-25:
emmm.It is a little complex.I want to test my method which is wrote in cpp. So I use lcm as a communication tool between the cpp code and python (client) code.Python code sent useful information to cpp code and cpp code use that information to calculate  the steer and throttle and send steer and throttle to cpp code.I also use multi-thread.After I run the python server,I run the python (client) code and then cpp code.The error below occur.I use version 0.7.0.

INFO: listening to server localhost:2000
CarlaClient connected
Received weather_id(1) and player_start(34) from channel(ENV_CONFIG)!
Starting new episode...
34

{'location': (56.26, 234.2, 0.22), 'agent_type': 'traffic_light', 'state': '2', 'orientation': (1.0728836059570312e-06, -0.9999999403953552, 0.0)}
{'location': (62.15, 219.92, 3.53), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'location': (77.74, 218.81, 3.55), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'location': (60.88, 244.82, 1.84), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'agent_type': 'player_agent', 'collision_vehicles': 0.0, 'forward_speed': 0.0, 'collision_pedestrians': 0.0, 'acceleration': (0.0, 0.0, 0.0), 'location': (71.0400390625, 237.43, 1.37), 'orientation': (-1.0, -6.4373016357421875e-06, 0.0), 'collision_other': 0.0, 'intersection_offroad': 0.0, 'intersection_otherlane': 0.0}
Received steer(0.0) , throttle(0.0) and on_inflexion(0) from channel(DECISION_INFO)!
send original decision
-5.231584964349167e-06
1.0

set autopilot succeed!
get info around succeed!
{'location': (56.26, 234.2, 0.22), 'agent_type': 'traffic_light', 'state': '2', 'orientation': (1.0728836059570312e-06, -0.9999999403953552, 0.0)}
{'location': (60.88, 244.82, 1.7528887939453126), 'agent_type': 'pedestrian', 'box_extent': (45.0, 35.0, 100.0), 'forward_speed': 0.0, 'orientation': (1.0, 0.0, 0.0)}
{'agent_type': 'player_agent', 'collision_vehicles': 0.0, 'forward_speed': -4.38001546143596e-09, 'collision_pedestrians': 0.0, 'acceleration': (6.57002274806473e-08, 4.229321699713434e-13, 0.0), 'location': (71.0400390625, 237.43, 1.3427870178222656), 'orientation': (-1.0, -6.4373016357421875e-06, 0.0), 'collision_other': 0.0, 'intersection_offroad': 0.0, 'intersection_otherlane': 0.0}

Done.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "client_example_for_shanbo.py", line 164, in send_measurement
    measurements,_ = client.read_data()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/client.py", line 117, in read_data
    data = self._stream_client.read()
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 74, in read
    header = self._read_n(4)
  File "/home/smielab/simulator/carla_0.7.0/PythonClient/carla/tcp.py", line 84, in _read_n
    raise TCPConnectionError(self._logprefix + 'not connected')
carla.tcp.TCPConnectionError: (localhost:2001) not connected

^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

My python(client) code like this:
#include"env_info/agent_info.hpp"
#include"env_info/around_agent_info.hpp"
#include"env_info/env_config.hpp"
#include"player_agent_behavior/decision_info.hpp"
#include<lcm/lcm-cpp.hpp>
#include<iostream>
#include<cstdlib>
#include<pthread.h>
#include<unistd.h>
#include<semaphore.h>

using namespace std;

#define RECEIVE_THREAD 0
#define SEND_THREAD 1

lcm::LCM env_config_lcm_send("udpm://239.255.76.63:7667?ttl=100");//send env_config message
lcm::LCM decision_info_lcm_send("udpm://239.255.76.63:7667?ttl=100");//send decision_info message
[... 571 lines cut ...]
if __name__ == '__main__':

    try:
        #initialize receive message
        env_config_msg = env_config()
        decision_info_msg = decision_info()
        on_inflexion = True
        #initialize lcm
        env_config_lcm_receive = lcm.LCM("udpm://239.255.76.63:7667?ttl=100")
        env_config_lcm_receive.subscribe("ENV_CONFIG",env_config_handler)
        around_agent_info_lcm_send = lcm.LCM("udpm://239.255.76.63:7667?ttl=100")
        decision_info_lcm_receive = lcm.LCM("udpm://239.255.76.63:7667?ttl=100")
        decision_info_lcm_receive.subscribe('DECISION_INFO',decision_info_handler)
        
        #other global variable
        autopilot_on = None
        autopilot_control = None
        
        #mutex
        on_inflexion_mutex = threading.Lock()
        client_mutex = threading.Lock()
        print('hehe')
        main()
    except KeyboardInterrupt:
        print('\nCancelled by user. Bye!')

nsubiron, 2018-01-25:
One thing that could be is the time-out in the server, by default the server closes the connection after 10 seconds of inactivity. It is possible that your client is taking more than 10 secs to reply, you can increase the time-out in the server's "CarlaSettings.ini"
```ini
[CARLA/Server]
; Time-out in milliseconds for the networking operations.
ServerTimeOut=10000
```
Then launch the server with the "-carla-settings" option

    $ ./Carla.sh -benchmark -fps=XX -carla-server -carla-settings="Path/To/CarlaSettings.ini"

nsubiron, 2018-01-25:
However in your case you might consider to bypass the Python client altogether and communicate with the server directly. _Do it only if you really know what you are doing_ though, and bear in mind that we are under heavy development and interface may change. Said that, the server-client protocol is described at [CarlaServer protocol](http://carla.readthedocs.io/en/latest/carla_server/#protocol) and the messages are serialized using [Google's Protocol buffers](https://developers.google.com/protocol-buffers/) as described in [carla_server.proto](https://github.com/carla-simulator/carla/blob/master/Util/Proto/carla_server.proto).

closed 2018-02-05 by carla-admin