carla-simulator/carla #170: can not Down load CARLA 0.7.1

Todspaniard, 2018-01-26:
Hello guys,
When I tried to  download  Compiled version of CARLA 0.7.1 or even previous version ,
then suddenly rejected  that  and shows  "sign up google and continue to google drive connection"  .
but can not resolved it after singed in.

That might happen  from 25th of Jan,2018.

nsubiron, 2018-01-26:
Hi @Todspaniard, thanks for the report. Yes, we are having issues with the download links, we are working on it, we hope they will be restablished soon.

Todspaniard, 2018-01-26:
Thanks a lot!

nsubiron, 2018-01-26:
The download link of the [latest release](https://github.com/carla-simulator/carla/releases/tag/0.7.1) should be working again :)

The previous releases are still unavailable, but should be available again in the coming days.

Todspaniard, 2018-01-26:
thanks for your quick action.
I confirmed  that i can download 0.7.1 compiled version.

I also can not  build CARLA project on linux  due to same issue and still struggling
with following error massage.

/CarlaUE4/Content_20180126213826'
Downloading Content.tar.gz...    32.77KB                                        Done.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

Its seems that  root case is same ,is'nt it.(google drive issue)

nsubiron, 2018-01-26:
Yes, that package took more time to upload again, it should be now fixed too

closed 2018-01-26 by Todspaniard