carla-simulator/carla #171: Simulation gets stuck

biboffff, 2018-01-26:
Hi,

I use Carla to create synthetic datasets by saving a couple of rendered images to disk.
I noticed, that the simulations gets stuck after the ~ 3000th - 5000th frame, causing the vehicle, the others vehicles and even pedestrians to standstill. I've attached an example image.

[image]

The images are rendered in 4k and the number of pedestrians and cars is randomly choosen between [30,80] and [100, 250], respectively.                  

Do you have any suggestion?

nsubiron, 2018-01-26:
Hi @biboffff,

Which method are you using for saving the images? Have you checked your RAM and disk usage?

biboffff, 2018-01-29:
Hi  @nsubiron,

thanks for replying.

I use the method save_to_disk, which is used in example_client.py
Disk and (gpu) RAM is fine.

nsubiron, 2018-01-29:
Hi @biboffff,

I tried to reproduce, and I noticed that pedestrians stop moving when you add so many. I'm suspecting this is an issue of Unreal's navigation system not handling very well a big number of agents (related #6). They spawn into the world and never move, we have seen before similar issues. We haven't tested what's the upper limit of pedestrians you can add, but I'm guessing around 80 it starts showing issues.

Vehicles too are not fail-proof, and if you add too many the risk of accident increases. Usually just a dead-lock, when two vehicles detect each other and never move again, currently the vehicle's AI does not handle this case. Then your car my gets stuck in a traffic jam that will never move.

I'm guessing the combination of the two is making the simulator look like is stuck.

Can you maybe try reducing the number of vehicles and pedestrians?

You could also start a new episode when you detect the car is not moving after certain time, if that's an option for you.

biboffff, 2018-01-30:
Hi @nsubiron,

thanks for your reply and for your effort.

closed 2018-01-30 by biboffff