carla-simulator/carla #172: FPS flag shows no effect

biboffff, 2018-01-26:
Hi,

I started the carla server through

`./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=160 -ResY=80 -i`

The variation of consecutive frames is very little since` fps = 15` (meaning that two consecutive frames show almost the identical scene) which I don't want to be stored on disk ( note the `-i` parameter). 

Unfortunately, setting `fps = 1` doesn't yield the desired effect neither on the images stored on disk nor in the viewer-window of carla. I also tried even less fps-steps, such as`fps=0.1` but without success.

So my current workaround is to store simply every 10th frame on disk, which is obviously very expensive and slow.. Shouldn't this be able to set with `-fps` parameter

I appreciate any suggestion.

nsubiron, 2018-01-26:
Hi @biboffff,

The `-benchmark -fps=15` is an Unreal engine feature not something we have implemented (see [UE4 command-line args](https://docs.unrealengine.com/latest/INT/Programming/Basics/CommandLineArguments/)). This value of FPS is clamped between 3 and 2000, you can see the current value in the HUD of the server window

[image]

biboffff, 2018-01-26:
Hello @nsubiron , thank you for the response

closed 2018-01-26 by biboffff