carla-simulator/carla #173: Post Processing

biboffff, 2018-01-26:
Hi,

according to http://carla.readthedocs.io/en/latest/cameras_and_sensors/ there are a couple of post-processing steps on the rendered images applied.

Is it possible to turn them off in the release version?

Best Regards,
Chris

nsubiron, 2018-01-26:
Hi, you can set your camera with `PostProcessing="None"`. This way you will retrieve the images with no post-processing at all.

biboffff, 2018-01-26:
Hi,

thanks for your response.
Is it also possible to set each post-process effect individually?

nsubiron, 2018-01-26:
No sorry, that's not possible in the release version. For that you would need to compile from source, remove the unwanted post-processing, and compile your own release version.

biboffff, 2018-01-29:
Alright, thank you!

biboffff, 2018-02-02:
Hello @nsubiron 

Setting the PostProcessing flag to 'None' doesn't only turn Vignette, Grain Jitter, Lense Flare, etc. off, but also rendering features, as one can see here: 
[image]

So my re-question is, if there is a way to turn the complete (not only single effects as I asked before) **image postprocessing** off?
This works without problems in the editor, but then the pedestrian model is missing. It would be good if it would also work in the release version.

Best regards,
chris

nsubiron, 2018-02-02:
Hi @biboffff,

We don't have any mechanism for doing this in the release since the post-process is packaged in the scene.

On the other hand, we are currently working on the new pedestrian models; they will be available in the build-from-source version too in a few weeks.

biboffff, 2018-02-02:
Alright, thanks!

closed 2018-02-02 by biboffff