carla-simulator/carla #175: [Segmentation] Different tags for traffic lights and Speed limit Traffic Sign

jokla, 2018-01-26:
Hi!

I wanted to try to automatically collect data (images and bounding boxes) for object detection. I was thinking to use the segmentation map to create the bounding boxes, but I noticed that the traffic light and the speed limit traffic sign have the same class. How can I change the class of the traffic lights? From Unreal?

Thank you very much!

nsubiron, 2018-01-26:
Hi @jokla,

Objects are tagged based on their file system path, specifically the folder name inside "Unreal/CarlaUE4/Content/Static/*".

You would need to split the "TrafficSigns" folder in two and move the objects accordingly. As long as you do this from within Unreal editor, Unreal will automatically make sure that links to those objects still work (probably using redirectors).

Then you would need to change the C++ code at "Tagger.h" and "Tagger.cpp". Add a new label to the `ECityObjectLabel` enum, and its corresponding check inside `GetLabelByFolderName()` function.

I think that should be it. Let us know if you have any issue.

Cheers!
Nestor

closed 2018-02-03 by nsubiron

MRAWAY77, 2018-05-31:
hi i follow the instruction above and modify the tagger.cpp portion into:

static ECityObjectLabel GetLabelByFolderName(const FString &String) {
  if      (String == "Buildings")       return ECityObjectLabel::Buildings;
  else if (String == "Fences")          return ECityObjectLabel::Fences;
  else if (String == "Pedestrians")     return ECityObjectLabel::Pedestrians;
  else if (String == "Pole")            return ECityObjectLabel::Poles;
  else if (String == "Props")           return ECityObjectLabel::Other;
  else if (String == "Road")            return ECityObjectLabel::Roads;
  else if (String == "RoadLines")       return ECityObjectLabel::RoadLines;
  else if (String == "SideWalk")        return ECityObjectLabel::Sidewalks;
  else if (String == "TrafficSigns")    return ECityObjectLabel::TrafficSigns;
  else if (String == "Vegetation")      return ECityObjectLabel::Vegetation;
  else if (String == "Mustang")         return ECityObjectLabel::Mustang;
  else if (String == "AudiTT")          return ECityObjectLabel::AudiTT;
  else if (String == "AudiA2_")         return ECityObjectLabel::AudiA2_;
  else if (String == "Walls")           return ECityObjectLabel::Walls;
  else                                  return ECityObjectLabel::None;
}

and the tagger.h into:
enum class ECityObjectLabel : uint8
{
  None         =   0u,
  Buildings    =   1u,
  Fences       =   2u,
  Other        =   3u,
  Pedestrians  =   4u,
  Poles        =   5u,
  RoadLines    =   6u,
  Roads        =   7u,
  Sidewalks    =   8u,
  TrafficSigns =  12u,
  Vegetation   =   9u,
  Mustang      =  10u,
  Walls        =  11u,
  AudiTT       =  12u,
  AudiA2_      =  13u,
};

I complied and run the file but i dont get Mustang AudiTT and AudiA2_ tagged when i zoom in the Seg Output pixels?

Did i miss out something?

MRAWAY77, 2018-05-31:
I had copy original file path of these 3 vehicles and paste it onto the same folder path as the rest of the labels./

nsubiron, 2018-05-31:
Unfortunately it only takes into account the first level folder inside "Content/Static", so you would need to create a new folder there and move the vehicles' meshes. (What it counts is the static mesh, not the vehicle blueprint).

But maybe it is easier to modify Tagger.cpp to recurse into folders, or check the full path in `GetLabelByPath()`.

MRAWAY77, 2018-06-01:
sorry but how do you modify Tagger.cpp to recurse into folders, or check the full path in GetLabelByPath()?

MRAWAY77, 2018-06-01:
how do you know which are the vehicle meshes for Mustang for example... i tried to move the static meshes but tags aint working

nsubiron, 2018-06-01:
```cpp
const FString Path = "/Game/Static/Vehicles/4Wheeled/Mustang";
TArray<FString> StringArray;
Path.ParseIntoArray(StringArray, TEXT("/"), false);
```
Here `StringArray` contains  `["Game", "Static", "Vehicles", "4Wheeled", "Mustang"]`. Then instead of checking only `StringArray[3]` you can check further folders too.

I think this is the best approach because there is not a standard on where to put the vehicles further than those folders I mention.

MRAWAY77, 2018-06-13:
why is it that i cant tag both mustang and Mini at the same time? but i could do it if its either one alone? 

static void SetStencilValue(
    UPrimitiveComponent &Component,
    const ECityObjectLabel &Label,
    const bool bSetRenderCustomDepth) {
                Component.SetCustomDepthStencilValue(CastEnum(Label));
                // Component.SetRenderCustomDepth(
                // bSetRenderCustomDepth &&
                // (Label == ECityObjectLabel::Mustang));
                Component.SetRenderCustomDepth(
                bSetRenderCustomDepth &&
                (Label == ECityObjectLabel::Mini));
                // Component.SetRenderCustomDepth(
                // bSetRenderCustomDepth &&
                // (Label == ECityObjectLabel::NissanPatrol));
                // Component.SetRenderCustomDepth(
                // bSetRenderCustomDepth &&
                // (Label == ECityObjectLabel::Bmwlsetta));
}

// =============================================================================
// -- static ATagger functions -------------------------------------------------
// =============================================================================

void ATagger::TagActor(const AActor &Actor, bool bTagForSemanticSegmentation)
{
#ifdef CARLA_TAGGER_EXTRA_LOG
  UE_LOG(LogCarla, Log, TEXT("Actor: %s"), *Actor.GetName());
#endif // CARLA_TAGGER_EXTRA_LOG

  // Iterate static meshes.
  TArray<UStaticMeshComponent *> StaticMeshComponents;
  Actor.GetComponents<UStaticMeshComponent>(StaticMeshComponents);
  for (UStaticMeshComponent *Component : StaticMeshComponents) {
    const auto Label = GetLabelByPath(Component->GetStaticMesh());
    if(Label == ECityObjectLabel::Mustang)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    if(Label == ECityObjectLabel::Mini)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    // if(Label == ECityObjectLabel::NissanPatrol)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    // if(Label == ECityObjectLabel::Bmwlsetta)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
#ifdef CARLA_TAGGER_EXTRA_LOG
    UE_LOG(LogCarla, Log, TEXT("  + StaticMeshComponent: %s"), *Component->GetName());
    UE_LOG(LogCarla, Log, TEXT("    - Label: \"%s\""), *GetLabelAsString(Label));
#endif // CARLA_TAGGER_EXTRA_LOG
  }

  // Iterate skeletal meshes.
  TArray<USkeletalMeshComponent *> SkeletalMeshComponents;
  Actor.GetComponents<USkeletalMeshComponent>(SkeletalMeshComponents);
  for (USkeletalMeshComponent *Component : SkeletalMeshComponents) {
    const auto Label = GetLabelByPath(Component->GetPhysicsAsset());
    if(Label == ECityObjectLabel::Mustang)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    if(Label == ECityObjectLabel::Mini)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    // if(Label == ECityObjectLabel::NissanPatrol)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);
    // if(Label == ECityObjectLabel::Bmwlsetta)   SetStencilValue(*Component, Label, bTagForSemanticSegmentation);

Naeem-996, 2018-08-30:
Hi @marcpuig, 
    I am running the carla server (version 0.8.2) within the Unreal Editor(4.18.2). Here are the things that I did:
1. Within the UE4 content browser, move only the "static mesh" of various vehicles and city lights and put them into the respective named folder in /Contents/Static/. My folder hierarchy looks as follows:
/Contents/Static/AudiTT/<static_mesh_for_AudiTT>
/Contents/Static/Citylamp/<Citylamp>
...
2. After that, I "Fix up redirectors in folder" for the "Static" folder to clean up the unreal folders and links.
3. Add those respective tags into the "Tagger.h" and "Tagger.cpp" like explained in the documentation. 
4. Compile and run the carla server within Unreal Editor.
5. Add the same label and color values into carla_0.8.2/carla/Image_converter.py > "labels_to_cityscapes_palette(image)" function.  
6. Run "manual_control.py" client. 
OBSERVATIONS:
The city lamps and other things which I wanted to label appeared black(at least different than before) in the semantic segmentation which means they are still labelled as "none". I have checked the folder names which match exactly. Could you please guide me if I made a mistake or missed something? 

Thank you so much.

SoumiDas, 2018-09-09:
How would I be able to distinguish between different traffic signs and type of speed sign programmatically?

It'd be really helpful to know!

Thanks!

commit 34331086 (2018-02-02): Close #175, note on adding new tags