carla-simulator/carla #176: MAC OS X Building

MFaisalZaki, 2018-01-26:
Hello Sir,

I'm working on building CARLA for MAC OSX but I'm facing this problem regarding downloading some file named content from a google drive, If you may send me a direct download link to be able to download myself then I'll try to automate the whole process soon.

I have attached terminal output when I executed the Update.sh script.
[image]

MFaisalZaki, 2018-01-27:
I think this occurs due to google drive permissions, So I have requested an access for this download content. If you may give me this permission, to be able to download the file myself, as for the python script download_from_gdrive.py the token received is always None, which aborts the download.

nsubiron, 2018-01-29:
Hi @MFaisalZaki,

We had some issues with the GDrive storage during Friday and the links went down. That's probably the cause of this. It should be fixed now. Have you tried again? Let us know if you still have issues with this.

MFaisalZaki, 2018-02-02:
I'm have compiled Carla successfully and generated Xcode project files but now I receive this error regarding permission denied for "touch: casetestABC: Permission denied"
[image]

MFaisalZaki, 2018-02-02:
Fixed the issue was I didn't have write permissions on the UE4 folder since I have installed it using Epic Launcher and didn't compile it.

nsubiron, 2018-02-02:
Great :) thanks for sharing, it seems like it can be a common issue.

MFaisalZaki, 2018-02-02:
Also I have this problem, when I clean Carla it deletes UE4Editor.app 
[image]

MFaisalZaki, 2018-02-02:
Done. I can run Carla on MAC OS but I can't find ./CarlaUE4.sh to run as a server.

Can you provide me with an example I don't want to download 2.0 GB to check a shell file.

MFaisalZaki, 2018-02-02:
I'm getting missing assets but the simulator is working, how this is possible 
[image]

nsubiron, 2018-02-03:
There is no _"CarlaUE4.sh"_ script in the source version of CARLA. Once you open the project in the Unreal Editor, you can hit Play to run CARLA.

The missing assets that is complaining about are Unreal's Automotive Materials, follow the instructions in the documentation to [Add Automotive Materials](http://carla.readthedocs.io/en/latest/how_to_add_automotive_materials/).

MFaisalZaki, 2018-02-06:
Here are the modifications for adding support for CARLA on mac os x. Sorry I didn't have time to automate it.
[mac_modifications.zip](https://github.com/carla-simulator/carla/files/1698547/mac_modifications.zip)

nsubiron, 2018-02-06:
Great! thanks for sharing. Do you think you could do a pull-request with the changes?

MFaisalZaki, 2018-02-06:
CARLA simulator is running slow in the editor, I have selected the quality to be low and the material also to be low and still the same issue. It runs at max 5 FPS !!!.

I thought at first it was due to the absence of -pthread flag, but after I added the flag no changes in the performance.

Please note that I'm running on Macbook pro with retina display with 8 cores i7 and 8 GB RAM with SSD disk.

I also received a notification saying since the Xcode installed is not 8.3.x then performance will be degraded, where my xcode version is 9.1

Please advice.

nsubiron, 2018-02-07:
> CARLA is a very performance demanding software, at the very minimum you would
need a computer with a dedicated GPU capable of running Unreal Engine. See
[Unreal Engine's recommended hardware](https://wiki.unrealengine.com/Recommended_Hardware).

CARLA requires a very powerful dedicated GPU to run smoothly. At the moment CARLA has significant performance issues, we are working hard to improve this, but it is something that takes a lot of time and effort.

See also the discussion about performance in #10.

MFaisalZaki, 2018-02-07:
Is it possible to have CARLA like OpenAI Gym environments, but the difference is that CARLA can provide both raw input images, processed input images or any set of states required by the user.

I'm working on motion planning algorithm using machine learning for my masters degree and I was thinking to modify CARLA to provide a set of states which will allow my algorithm to work as expected.

What do you think ?

Also if you can provide a way to remove the attached camera from the vehicle ? if doing so will help in speeding up the FPS ?!

closed 2018-03-07 by nsubiron

amritas02, 2018-05-18:
Can anyone guide me what to do next? I've built Unreal Engine with Xcode but don't know how to proceed with building Carla!

MFaisalZaki, 2018-05-18:
Merge the changes found in ticke 176 and everything will work.

amritas02, 2018-05-21:
I did all the steps mentioned above. But when I try to open CarlaUE4.uproject with Unreal Engine, it gives me this error. FYI, I have built my Unreal Engine from the source manually. 

[image]

When I say yes to the above, it tries to build the missing files and then says build failed. 

I have been over this so many times - it's not working.

Abanoub-G, 2018-09-29:
> I did all the steps mentioned above. But when I try to open CarlaUE4.uproject with Unreal Engine, it gives me this error. FYI, I have built my Unreal Engine from the source manually.
>
> [image]
>
> When [...]

Hi,
    Did you manage to solve this problem?

Abanoub-G, 2018-09-29:
> Done. I can run Carla on MAC OS but I can't find ./CarlaUE4.sh to run as a server.
> 
> Can you provide me with an example I don't want to download 2.0 GB to check a shell file.

Hey can you please tell me what you exactly did? I have reached to the point where I type on terminal "make launch" and I get this in return:

clang 5.0 is required, but it's not installed.
make sure you build Unreal Engine with clang 5.0 too.
make: *** [setup] Error 1

I already have clang version 7.0.0 installed. I have also installed version 5.0.0 but I don't know how to make my Mac use the 5.0.0 version instead of the 7.0.0 version. Any idea how I can go about this.

Also I don't know how I can use the files you modified below...

> Here are the modifications for adding support for CARLA on mac os x. Sorry I didn't have time to automate it.
> [mac_modifications.zip](https://github.com/carla-simulator/carla/files/1698547/mac_modifications.zip)
> 
> PS: Please check it and if you may integrate it and you can close this issue.

If you can show me how your files can be used that would be awesome. All what I did so far was by following the guide on how to build Carla on windows.

Thanks

amritas02, 2018-10-03:
> Hi,
> Did you manage to solve this problem?

No I didn't. I just stopped using it