carla-simulator/carla #177: TCP connection always breaks

18605973470, 2018-01-27:
I plan to write a reinforcement learning environment with carla.
But I find that tcp connection is easy to break
 ( e.g. "read_data()" failed or "connect()" failed in File "carla/tcp.py")
It does great harm to the robustness of my environment.

So what should I do to avoid this ?

18605973470, 2018-01-27:
Sometimes the error is 
"carla.tcp.TCPConnectionError: (127.0.0.1:2000) failed to connect: [Errno 111] Connection refused"
But I am sure that server is on

nsubiron, 2018-01-29:
Hi @18605973470,

There are a number of reasons why the connection might fail, server timed-out, server is doing a reset. In principle it should be fine to catch this exception and try again.

See also #169.

With release 0.7.1 has been a fix related to this in Python 2, you may try with the latest client code to see if it fixes this.

18605973470, 2018-01-31:
I don't know how the server behaves.
But connection error still occurred when I tried to reconnect to server right after I used "disconnect"(or an exception was captured).
Maybe "SO_REUSEADDR" flag(or something similar) should be used in server TCP socket?

nsubiron, 2018-02-01:
Hi @18605973470,

We don't have a mechanism to recover from a disconnect in the middle of an episode, so the server needs to abandon current episode and start a new one. This takes time, it has to reload all settings and set up a new scene. There is not much that can be done but waiting a few seconds until Unreal is ready again.

closed 2018-02-12 by nsubiron