carla-simulator/carla #178: Unreal Engine git

tartavull, 2018-01-28:
I couldn't clone this https://github.com/EpicGames/UnrealEngine.git

encore2020, 2018-01-28:
me too, then how to download?

CapitalZe, 2018-01-28:
You need to sign up an account on the UE website and then link this to your Github account, you will be prompted to enter your Github username and pword in terminal after you exact git clone.

Frankly, it's not clear but easy when you know what to do.

nsubiron, 2018-01-29:
I'm making this clearer in the docs

closed 2018-02-01 by carla-admin

ahrnbom, 2021-07-15:
Even after creating an UE account, and linking it with my github account, and then trying to clone UnrealEngine (either from EpicGames or the CARLA-fork) just gives a "fatal: repository not found" error. And yes, I am sure that I am using the same github account for both these things

lraulc, 2022-04-29:
I know this is far into the future from when you posted, but in order to be able to clone from terminal (2022), I had to do couple of steps:

1. Link Github account to Epic Games Account
2. Fork repository
3. Create an SSH key for your github account.
[Generating a New SSH - Github](https://docs.github.com/es/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent)
4. After generating you SSH, assign it to your account:
[Connecting Github with SSH](https://docs.github.com/es/authentication/connecting-to-github-with-ssh/adding-a-new-ssh-key-to-your-github-account)
5. Proced to clone using SSH instead of HTTPS
6. Voilá.
Hope this helps!

Aimo2027, 2024-11-13:
I am facing these issues using this link (git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git)
How to fix it?

(Cloning into 'UnrealEngine'...
remote: Enumerating objects: 111593, done.
remote: Counting objects: 100% (111593/111593), done.
remote: Compressing objects: 100% (92142/92142), done.
error: RPC failed; curl 92 HTTP/2 stream 7 was not closed cleanly: CANCEL (err 8)
error: 4718 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output)

colibri17, 2024-12-04:
> Even after creating an UE account, and linking it with my github account, and then trying to clone UnrealEngine (either from EpicGames or the CARLA-fork) just gives a "fatal: repository not found" error. And yes, I am sure that I am using the same github account for both these things

I am in the same situation. Any help?

xADawgx, 2025-01-01:
> I am facing these issues using this link (git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git) How to fix it?
>
> (Cloning into 'UnrealEngine'... remote: Enumerating objects: [...]

I'm having a similar error as well:

Cloning into 'X:\Game Content\UnrealEngineSource\UnrealEngine'...
remote: Enumerating objects: 6511107, done.        
remote: Counting objects: 100% (162/162), done.        
remote: Compressing objects: 100% (47/47), done.        
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
error: 7581 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

I am downloading from the app using the url method

duongtran11, 2025-02-16:
> I'm having a similar error as well:
>
> Cloning into 'X:\Game Content\UnrealEngineSource\UnrealEngine'... remote: Enumerating objects: 6511107, done. remote: Counting objects: 100% (162/162), done. [...]

I'm getting the same error
On windows 10 pc