carla-simulator/carla #179: image converter to create  senmatic seg image from disk saved data

Todspaniard, 2018-01-28:


Todspaniard, 2018-01-28:
Hi guys.
I would like to make  image converter based on  instruction as bellow.
after cd  to Carla/Util/ImageConverter make it, following error message come up.

    main.cpp:13:10: fatal error: 'boost/filesystem.hpp' file not found
    #include <boost/filesystem.hpp>
             ^
    1 error generated.
     Makefile:11: recipe for target 'release' failed
     make: *** [release] Error 1

could you suggest what is wrong and how to resolve it.

I try  export  CPLUS_INCLUDE_PATH="$C_PLUS_INCLUDE_PATH  ...  "  
but situation is same.

thanks

nsubiron, 2018-01-29:
Hi @Todspaniard,

You need to install before the dependencies mentioned in the ImageConverter's README

> libboost_system, libboost_filesystem, libboost_program_options, libpng, libtiff,
libjpeg and libomp.

In Ubuntu you should be able to find them with `apt-get`.

Todspaniard, 2018-01-29:
Thanks,
I did it and confirmed ok.

closed 2018-01-29 by Todspaniard

raviks789, 2018-02-22:
Hi,

I am trying to make image converter in windows and I get the same error as mentioned initially even though I have installed boost and also added its path in the environment variables.

Can anybody help with this issue?

Thanks in advance

nsubiron, 2018-02-22:
Hi @raviks789,

Thanks for the report. Yes, the Makefile is only valid for Linux, I will add this to the list of things to do for adding Windows support (#241). Meanwhile you could try building it either adding the files in that folder to a Visual Studio project, or using some Linux utils like Cygwin.

raviks789, 2018-02-22:
Hi @nsubiron,

Thank you for your reply. I already tried doing with both Cygwin and Msys, I still encountered the same issue. I tried creating a project in visual studio and building them, but I got some 24 errors.I have attached the error log, kindly help resolving them.

Thank you in advance.
[Error_log.txt](https://github.com/carla-simulator/carla/files/1747642/Error_log.txt)

nsubiron, 2018-02-22:
@raviks789 Sorry I don't have a Windows machine to test it right now, but it seems that you don't have C++14 enabled, you need to configure Visual Studio to use C++14 and link against the dependencies.

I still think the easiest way would be to use Cygwin, but you need to install the dependencies inside Cygwin so it will be able to link them. Usually you can select them when running the Cygwin setup.

raviks789, 2018-02-22:
@nsubiron I installed all the required packages using Cygwin and tried doing make. But still encountered the same error.

 make
clang++ -Wall -Wextra -std=c++14 -fopenmp -O3 -DNDEBUG -o bin/image_converter main.cpp -lboost_system -lboost_filesystem -lboost_program_options -lpng -ljpeg -ltiff
main.cpp:14:10: fatal error: 'boost/filesystem.hpp' file not found
#include <boost/filesystem.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile:12: release] Error 1

komms, 2018-07-24:
@raviks789  did you solve that error?

nsubiron, 2018-07-25:
It seems like an issue with the cygwin installation of boost, try to locate the boost header files and add that folder to the include path.