carla-simulator/carla #183: Rendering Issue

biboffff, 2018-01-30:
Hello,

When I'm using the carla-built-from-source version (texture streaming is disabled), all lanterns appear bugged. This doesn't happen using the release version. 

[image]

Do you have an idea what could go wrong?

Thanks in advance,
Chris

TheNihilisticRobot, 2018-01-31:
HI @biboffff
 
So, i'm not actually sure what the problem you are perceiving is. Is it the difference between the post on the background and the one of the foreground? if so, what you are seeing is the difference between LODs, a system to reduce rendering costs reducing polygons the further away an object is.

If that's not it i'll need a bit more info:
What exactly makes you think the post is bugged.
Different screenshots of the release and built from source versions.

biboffff, 2018-01-31:
Hi @TheNihilisticRobot 

thanks for your reply.

I mean those grey-black dots and bumps on the lantern. I just marked to lantern in the background as well, because if you zoom in, it is there also present.

I thought this doesn't appear in the release version, but meanwhile I noticed it there as well.
Anyway, here is an example how I thought that each lantern should look like:

[image]

Do you know now what I try to point out?

Thanks!

chris

TheNihilisticRobot, 2018-02-01:
Ah! Those bumps are there intentionally. We made our props a bit rusty to dissimulate the low poly assets and give a more organic look. 
You are free to modify the model material if you will. This particular one is located in Content/Static/Pole/Prop_CityLamp and is called CityLamp_Body_Mat. Just remove the Normal Map from the material and you'll get a clean flat post.

biboffff, 2018-02-01:
Alright, thank you very much. I just have one more question:

I need to adjust the post processing pipeline, e.g. by turning bloom or lense flare off.

I think the easiest way to achieve this, is to built carla from source and to do all adjustments in the unreal editor. Afterwards, I could run it as server, connect with a client and all is fine.

Can you tell me if this approach works without problems or would you recommend to do all adjustments in the *ini files, followed by a rebuild?

Thanks
chris

TheNihilisticRobot, 2018-02-02:
Yeah, that would definitely work, this way you can also test the post process while you change them in the editor, making it easier for you to get the desired results.

biboffff, 2018-02-02:
alright thanks alot!

closed 2018-02-02 by biboffff