carla-simulator/carla #184: Fine control of NPC (make the NPC controllable)

dehabu, 2018-01-30:
By request of Alexander Anderson <aga@berkeley.edu>

Extending the current API to control the rest of dynamic objects. At least to have a finer control of their initialization:

Use cases: 
- creating situations in which we have a car in front of us blocking our car, etc. 
- create "events", e.g. car overtaking another car, etc.

sandman, 2018-02-14:
When is this feature planned to be released?

felipecode, 2018-02-14:
Hey @sandman 

This is a high priority feature for us. But we have planned date to release yet.

Thanks for your interest.

Cheers.

ciumonk, 2018-04-14:
I've been working on this for a while now. Here what works:
-spawning a pedestrian(or more) with a certain trajectory and speed, configurable from the settings file or sent from the python client(to be used in crash simulations). This basically uses a combination of SetActorLocation and Controller->MoveTo along with selectively disabling the part of code that stops the pedestrian if he is about to intersect with a vehicle.
-the same with cars

My only issue is that pedestrians seem to remain stuck around intersections sometimes. I haven't been able to find any code that would affect this. I have a suspicion that it might be somewhere in a blueprint, any ideas? Pedestrians spawned on the sidewalk in the vicinity of intersections sometimes never move using Controller->MoveTo, unless I interpolate using SetActorLocation(start+ratio*end)....

ciumonk, 2018-04-18:
Hi, I have this partly working. Perhaps I could submit a PR after the ITSC2018 deadline(1 May)...

anthonypan08, 2018-05-24:
Any updates on when these features will be included?

nsubiron, 2018-10-24:
Hey, this was already possible in 0.9.0, and will be significantly improved in 0.9.1.

closed 2018-10-24 by nsubiron

peiyunh, 2020-09-04:
> Hey, this was already possible in 0.9.0, and will be significantly improved in 0.9.1.

Hi @nsubiron could you kindly point me to an example for controlling NPCs? Thanks!